Moddable-OpenSource/moddable #988: xsbug locate source has bad .js filter on Linux

tve, 2022-12-15:
**Build environment:** Arch Linux

**Description**
When xsbug can't locate a source file due to directory path mismatch since compilation it has a "locate" button which brings up a file browser/selector. This selector ostensibly filters for .js files but that appears broken.

**Steps to Reproduce**
1. Build an example project using docker, which causes paths to differ from the host
2. Start xsbug and load the app, e.g. esp32
3. xsbug comes up and tries to show /data/main.js, there's a locate button, use that and try to find the file

Here's xsbug:
[image]

So far so good, the locate button brings up this file dialog, which I have open on the source directory:
[image]

Notes:
- `ls` in that dir shows: `foo.js  js  main.js  manifest.json`
- I created `js` and `foo.js` after seeing that `main.js` doesn't appear
- the filter at the bottom-right says `js`, I wonder whether it should say `.js`, it's ominous that the only file shown is the one called exactly `js`...
- I assume you can create the issue by moving the source files after building the image

phoddie, 2022-12-15:
> Build an example project using docker, which causes paths to differ from the host
>
> I assume you can create the issue by moving the source files after building the image

Please provide steps to reliably  reproduce the issue without docker. Thank you.

tve, 2022-12-15:
STR:
- Build a project that produces an error in xsbug, for example the drivers/lora example
- xsbug shows the error at the bottom (`/home/sw/moddable/moddable/examples/drivers/lora/main.js (26) # Break: main: get device: undefined variable!`) and the source file
- Rename trhe source directory, e.g., `mv lora lora2`
- the source file immediately disappears from xsbug and you get a "locate" button
- hit the button a file browser shows up, navigate to the `lora2` dir
- the source file `main.js` is not visible in the file browser

phoddie, 2022-12-17:
Thank you for the updated steps. We'll take a look.

phoddie, 2022-12-27:
This has been fixed by [this commit](https://github.com/Moddable-OpenSource/moddable/commit/86e1a62527bad2306991b822509c1cdaf8634254). We had previously disabled the file filter on Windows and macOS. That's now been appleid to Linux as well.

closed 2022-12-27 by phoddie