Moddable-OpenSource/moddable #990: minor: new sx1276 version in lora driver (and it's SX not ST)

tve, 2022-12-15:
The SX127x LoRa driver is misnamed. LoRa technology & chips are produced by Semtech, not ST Microelectronics, their chips tend to have an "sx" prefix.
In the initialization the st127x.js driver checks for 0x12 in the version register of the chip:
https://github.com/Moddable-OpenSource/moddable/blob/public/modules/drivers/st127x/st127x.js#L178

Per https://forum.lora-developers.semtech.com/t/new-sx1276-chip-revision-with-reg-version-0x13/588 there are now sx1276 chips with `version == 0x13` so you may want to add that to the check. I haven't bumped into any such chip, but mine are mostly fairly old...

phoddie, 2022-12-15:
Thank you for the report. This seems like a great first PR. Please see our [Contributing](https://github.com/Moddable-OpenSource/moddable/blob/public/contributing.md#submitting-pull-requests) document.

tve, 2022-12-15:
How do you suggest fixing the s/ST/SX/ issue, e.g. backwards compat.?

phoddie, 2022-12-15:
I'm not sure what you mean by "backwards compat." You mentioned two issues above:

1. The naming is incorrect.
2. The version check should accept both 0x12 and 0x13

tve, 2022-12-15:
The version check can obviously just be added as an OR.
Renaming the driver to SX127x will break existing uses. Uses within the SDK can obviously be fixed at the same time, but that still leaves external ones. I suppose a st127x.js proxy can be left, I believe there's nothing being exported that has "ST" in it.

phoddie, 2022-12-15:
I see. By "backwards compat" you are referring to the impact of changing file names. Thank you.

I think in this case it is fine to simply rename the files and update references inside the Moddable SDK. It is easy enough to update projects to use the new name. We can note that in the release notes

phoddie, 2022-12-27:
I've made these changes. They will be available in the coming days. Thanks again for the report!

FWIW – the lora and lora_ui examples didn't require any changes, even though the file and path names changed. That's because they access the LoRa driver through the device global. Since the `esp32/heltec_lora_32` manifest was updated with the new paths, the examples continued to work as before.

closed 2023-01-10 by phoddie