Moddable-OpenSource/moddable #994: Socket error: -1 on MQTT Connection

diogoviannaaraujo, 2022-12-16:
**Build environment:** macOS
**Target device:** ESP32

**Description**
When trying to connect to AWS IoT services using custom authenticator connection fails with error -1

**Steps to Reproduce**
Created a custom authenticator on AWS IoT service
Verified it working by using mosquitto client:
```
mosquitto_pub -p 443 -i 'clientid1234' -h 'xxxxxxx-ats.iot.us-east-1.amazonaws.com' --cafile AmazonRootCA1.pem --tls-alpn mqtt -u 'username?x-amz-customauthorizer-name=MyCustomAuthorizer' -P 'MTIzNA==' -t customauthtesting -m "Hello World!"
```
It connects and publish a message successfully.

But when trying to connect on moldable Matt client using:
```
    connection = new MQTT({
        host: "xxxxxxxxx-ats.iot.us-east-1.amazonaws.com",
        id: "clientid1234",
        username: "username?x-amz-customauthorizer-name=MyCustomAuthorizer",
        password: "MTIzNA==",
        port: 443,
        Socket: SecureSocket,
        timeout: 60000,
        secure: {
          cache: false,
          applicationLayerProtocolNegotiation: "mqtt",
          protocolVersion: 0x303,
          certificate: new Resource("AmazonRootCA1.der"),
          verify: true,
          trace: true,
        },
      })
```

I get an error
`MQTT FAIL: socket error -1`

I can provide the complete host url if needed for testing

**Expected behavior**
Expect to connect and/or give a more meaningful error code

phoddie, 2022-12-16:
The TLS connection succeeded. That's visible in your trace `FINISHED: RSA-AES-CBC-SHA1`.

The remote endpoint disconnects almost immediately. The MQTT login is the first operation after the TLS connection is established. Comparing your call to the constructor with the [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#constructordictionary-8) shows a possibly relevant difference. You are passing the user name with the `username` property but the MQTT implementation expects it on the `user` property.

> Expect to connect and/or give a more meaningful error code

Understand that TLS seldom gives meaningful error messages. This seems to be a practice to avoid giving an attacker information that might help them.

diogoviannaaraujo, 2022-12-16:
Hmm got it, so it's not a TLS issue. 
The meaningful error code was only to fill something on the expected behavior, I understand that is not possible sometimes.

The user one was a good catch, switched and now im getting the disconnection after the onReady, so that problem is fixed, now there is another one, probably some auth config.

Thanks again phoddie!

closed 2022-12-16 by diogoviannaaraujo