Moddable-OpenSource/moddable #996: [Win10] esp32c3 fails to build

Sineos, 2022-12-19:
I would like to build for an esp32c3. It fails with
(all relevant environment information should be in the first few lines)

```
Starting build process...
Host system check: Windows 10 Pro
MCU Build system check: p1.0.0-rc.1 + #8ca3f6c @ m3.5.1-57-g7be4b23 (x86)
HOME directory check: C:\Users\souls
Creating build environment for platform esp32/esp32c3.
Working directory: \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
$IDF_PATH is defined: C:\Espressif\frameworks\esp-idf-v4.4.3
$IDF_TOOLS_PATH is defined: C:\Espressif
$IDF_PYTHON_ENV_PATH is not defined.
> cd \Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n
Creating build batch file...
> cmd.exe build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat" 
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.0
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.3

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.24.202208181.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.3\tools
    
Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.3\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

FC: Kann C:\USERS\SOULS\PROJECTS\MODDABLE\BUILD\TMP\ESP32\ESP32C3\RELEASE\B33DC2XT18N\XSPROJ-ESP32C3\PARTITIONS.CSV nicht ffnen - Datei oder Ordner nicht vorhanden

        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
Reconfiguring ESP-IDF...
warning: user value 1536 on the int symbol FREERTOS_ISR_STACKSIZE (defined at C:/Espressif/frameworks/esp-idf-v4.4.3/components/freertos/Kconfig:223) ignored due to being outside the active range ([2096, 32768]) -- falling back on defaults
Loading defaults file C:/Users/souls/Projects/moddable/build/tmp/esp32/esp32c3/release/b33dc2xt18n/xsProj-esp32c3/sdkconfig.mc...
-- Configuring done
-- Generating done
-- Build files have been written to: C:/Users/souls/Projects/moddable/build/tmp/esp32/esp32c3/release/b33dc2xt18n/xsProj-esp32c3/build
Executing action: reconfigure
Running cmake in directory c:\users\souls\projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n\xsproj-esp32c3\build
[... 385 lines cut ...]
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:388:26: error: incompatible types when assigning to type 'volatile union <anonymous>' from type 'int'
         GPIO.enable_w1ts = (0x1 << gpio_num);
                          ^
C:\Users\souls\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.c:392:14: error: 'gpio_dev_t' {aka 'volatile struct gpio_dev_s'} has no member named 'enable1_w1ts'; did you mean 'enable_w1ts'?
         GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
              ^~~~~~~~~~~~
              enable_w1ts
NMAKE : fatal error U1077: 'C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin\riscv32-esp-elf-gcc.EXE' : return code '0x1'
Stop.
```

`C:\Users\souls\Projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n`:
[b33dc2xt18n.zip](https://github.com/ralphwetzel/node-red-mcu-plugin/files/10261982/b33dc2xt18n.zip)

This build is from within node-red but it makes no difference if I call the `build.bat`directly in `ESP-IDF 4.4 CMD`

```
CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"
pushd %IDF_PATH%
CALL "C:\Espressif\idf_cmd_init.bat"
popd
mcconfig -m -p esp32/esp32c3 ssid="aaa" password="xxx"
```

Any pointer would be appreciated. If you need more / other information, just let me know

phoddie, 2022-12-19:
Looks like the ESP-IDF defines hardware GPIO access differently on ESP32C3 from other targets. I don't have an ESP32C3 with me to try, but it looks straightforward to resolve the build error.

In owb_rmt.c, try replacing lines 386-393 with:

```c
    // attach GPIO to previous pin
#if kCPUESP32C3
        GPIO.enable_w1ts.val = (0x1 << gpio_num);
#else
    if (gpio_num < 32)
    {
        GPIO.enable_w1ts = (0x1 << gpio_num);
    }
    else
    {
        GPIO.enable1_w1ts.data = (0x1 << (gpio_num - 32));
    }
#endif
```

You'll also need to add this to the includes:

```c
#include "xsHost.h"
```

Please let me know how it goes. Thanks!

mkellner, 2022-12-19:
Also, please use ESP IDF v4.4.2 with esp32c3.

Sineos, 2022-12-20:
Many thanks @phoddie. It worked for compiling but then after:

```
Hash of data verified.
Compressed 3072 bytes to 123...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (123 compressed) at 0x00008000 in 0.1 seconds (effective 327.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Detecting chip type... ESP32-C3
Running ninja in directory c:\users\souls\projects\moddable\build\tmp\esp32\esp32c3\release\b33dc2xt18n\xsproj-esp32c3\build
Executing "ninja flash"...
Done
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
        1 Datei(en) kopiert.
Executing action: monitor
Serial port COM6
Connecting...--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x48
load:0x403cc710,len:0x67c
load:0x403ce710,len:0x2260
entry 0x403cc710
Core dump started (further output muted)
Received   1 kB...
Received   2 kB...
Received   3 kB...
Received   4 kB...
Received   5 kB...
Received   6 kB...

Core dump finished!
Exception in thread Thread-1:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
Exception in thread Thread-2:
Traceback (most recent call last):
  File "threading.py", line 932, in _bootstrap_inner
  File "threading.py", line 870, in run
  File "threading.py", line 870, in run
  File "subprocess.py", line 1366, in _readerthread
  File "subprocess.py", line 1366, in _readerthread
OSError: [Errno 22] Invalid argument
OSError: [Errno 22] Invalid argument
WARNING: Attempt to terminate the GDB process failed, because it is already terminated. Skip
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3fc9e1f4, name: 'main', GDB name: 'process 1070195188'

================== CURRENT THREAD REGISTERS ===================
ra             0x42005f52	0x42005f52 <modRunMachineSetup+500>
sp             0x3fc9ddd0	0x3fc9ddd0
gp             0x3fc91c00	0x3fc91c00 <country_info_24ghz+224>
tp             0x3fc22920	0x3fc22920
t0             0x4005890e	1074104590
t1             0x420324a4	1107502244
t2             0x600	1536
fp             0x3c1171d8	0x3c1171d8 <gxHeap+79536>
[... 74 lines cut ...]
.rtc.force_fast 0x50000010 0x0 RW  
.rtc.data 0x50000010 0x10 RW A
.rtc_noinit 0x50000020 0x0 RW  
.rtc.force_slow 0x50000020 0x0 RW  
.iram0.text 0x40380000 0x112c0 RWXA
.dram0.data 0x3fc91400 0x2270 RW A
.noinit 0x3fc93670 0x0 RW  
.flash.text 0x42000020 0xc825a R XA
.flash.appdesc 0x3c0d0020 0x100 R  A
.flash.rodata 0x3c0d0120 0x7b8c0 RW A
.eh_frame 0x3c14b9e0 0x708 R  A
.iram0.data 0x40391400 0x0 RW  
.iram0.bss 0x40391400 0x0 RW  
.dram0.heap_start 0x3fc97d00 0x0 RW  
.coredump.tasks.data 0x3fc9e1f4 0x154 RW 
.coredump.tasks.data 0x3fc9dd30 0x4b0 RW 
.coredump.tasks.data 0x3fc9ae5c 0x154 RW 
.coredump.tasks.data 0x3fc9ad40 0x110 RW 
.coredump.tasks.data 0x3fc9b5b8 0x154 RW 
.coredump.tasks.data 0x3fc9b510 0xa0 RW 
.coredump.tasks.data 0x3fc99cc0 0x154 RW 
.coredump.tasks.data 0x3fc99ba0 0x110 RW 

===================== ESP32 CORE DUMP END =====================
```

phoddie, 2022-12-20:
> It worked for compiling...

Cool. One problem solved.

> ... but then after..

Yes, I think that's why @mkellner suggested using ESP-IDF 4.4.2. There seems to be an incompatibility with ESP-IDF 4.4.3 and our ESP32C3 support that causes a crash on launch. We haven't had a chance to track that down yet.

Sineos, 2022-12-21:
> Yes, I think that's why @mkellner suggested using ESP-IDF 4.4.2.

```
Microsoft Windows [Version 10.0.19045.2364]
(c) Microsoft Corporation. Alle Rechte vorbehalten.

C:\Users\souls>cd c:\Espressif\frameworks\esp-idf-v4.4.2

c:\Espressif\frameworks\esp-idf-v4.4.2>install.bat
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, cmake, openocd-esp32, mconf, ninja, idf-exe, ccache
Skipping xtensa-esp32-elf@1.22.0-80-g6c4433a5-5.2.0 (already installed)
Skipping esp32ulp-elf@2.28.51.20170517 (already installed)
Skipping cmake@3.13.4 (already installed)
Skipping openocd-esp32@v0.10.0-esp32-20190313 (already installed)
Skipping mconf@v4.6.0.0-idf-20190628 (already installed)
Skipping ninja@1.9.0 (already installed)
Skipping idf-exe@1.0.1 (already installed)
Skipping ccache@3.7 (already installed)
Setting up Python environment
Installing Python packages from c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt
Requirement already satisfied: setuptools in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from -r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 4)) (65.6.3)
Requirement already satisfied: pyserial>=3.0 in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from -r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 8)) (3.5)
Requirement already satisfied: future>=0.15.2 in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from -r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 9)) (0.18.2)
Requirement already satisfied: cryptography>=2.1.4 in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from -r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 10)) (38.0.4)
Requirement already satisfied: pyparsing<2.4.0,>=2.0.3 in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from -r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 11)) (2.3.1)
Requirement already satisfied: cffi>=1.12 in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from cryptography>=2.1.4->-r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 10)) (1.15.1)
Requirement already satisfied: pycparser in c:\espressif\python_env\idf3.3_py3.10_env\lib\site-packages (from cffi>=1.12->cryptography>=2.1.4->-r c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt (line 10)) (2.21)
All done! You can now run:
   export.bat
c:\Espressif\frameworks\esp-idf-v4.4.2>export.bat
Setting IDF_PATH: c:\Espressif\frameworks\esp-idf-v4.4.2

Adding ESP-IDF tools to PATH...
    C:\Espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binutils\bin
    C:\Espressif\tools\cmake\3.13.4\bin
    C:\Espressif\tools\openocd-esp32\v0.10.0-esp32-20190313\openocd-esp32\bin
    C:\Espressif\tools\mconf\v4.6.0.0-idf-20190628\
    C:\Espressif\tools\ninja\1.9.0\
    C:\Espressif\tools\idf-exe\1.0.1\
    C:\Espressif\tools\ccache\3.7\
    C:\Espressif\python_env\idf3.3_py3.10_env\Scripts
    c:\Espressif\frameworks\esp-idf-v4.4.2\tools

Checking if Python packages are up to date...
Python requirements from c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

c:\Espressif\frameworks\esp-idf-v4.4.2>cd C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>build.bat

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"
**********************************************************************
** Visual Studio 2022 Developer Command Prompt v17.4.3
** Copyright (c) 2022 Microsoft Corporation
**********************************************************************
[vcvarsall.bat] Environment initialized for: 'x86'
Setting PYTHONNOUSERSITE, was not set
Using Python in c:\Espressif\frameworks\esp-idf-v4.4.2\
Der Befehl "null" ist entweder falsch geschrieben oder
konnte nicht gefunden werden.
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: c:\Espressif\frameworks\esp-idf-v4.4.2

Adding ESP-IDF tools to PATH...
No directories added to PATH:

c:\Espressif\frameworks\esp-idf-v4.4.2\;C:\Espressif\tools\idf-git\2.34.2\cmd\;C:\Espressif;C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.34.31933\bin\HostX86\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\VC\VCPackages;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TestWindow;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\CommonExtensions\Microsoft\TeamFoundation\Team Explorer;C:\Program Files\Microsoft Visual Studio\2022\Community\MSBuild\Current\bin\Roslyn;C:\Program Files\Microsoft Visual Studio\2022\Community\Team Tools\Performance Tools;C:\Program Files (x86)\Windows Kits\10\bin\10.0.19041.0\\x86;C:\Program Files (x86)\Windows Kits\10\bin\\x86;C:\Program Files\Microsoft Visual Studio\2022\Community\\MSBuild\Current\Bin\amd64;C:\Windows\Microsoft.NET\Framework\v4.0.30319;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\IDE\;C:\Program Files\Microsoft Visual Studio\2022\Community\Common7\Tools\;C:\Espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.28.51.20170517\esp32ulp-elf-binuti [... cut ...]
Python requirements from c:\Espressif\frameworks\esp-idf-v4.4.2\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\Users\souls\.node-red\mcu-plugin-cache\b33dc2xt18n>*** Update required to ESP-IDF v4.4.3
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md"Expected ESP IDF v4.4.3, found v3.3.1"
NMAKE : fatal error U1077: 'python' : return code '0x1'
Stop.
```

Seems I cannot get it to run under 4.4.2. Any pointer to what I have missed?

phoddie, 2022-12-27:
It looks like the ESP-IDF version check is incorrect on Windows. When I do the same (build for `esp32/esp32c3` with ESP-IDF v4.4.2 installed), there is a recommendation to update but the build continues:

```
# Reconfiguring ESP-IDF...
Recommend using ESP-IDF v4.4.3 (found v4.4.2)
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md
```

Let's try modifying the `versionCheck.py` script to do two things:

1. Print out the expected and found versions of ESP-IDF to see why it is mismatching
2. Never generate an error on version mismatch to see if that allows your build to progress

<details>
<summary>Here's a modified version of the script at `$MODDABLE/build/devices/esp32/xsProj-esp32c3/versionCheck.py`</summary>

```py
import sys

if len(sys.argv) < 3:
	print("Not enough parameters")
	sys.exit(1)

update = "  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md"
expected = sys.argv[1]
given = sys.argv[2]

g = given.split(".")
if len(g) < 3:
	g.append(0)
e = expected.split(".")
if len(e) < 3:
	e.append(0)

print("EXPECTED ESP-IDF " + expected + " (FOUND " + given + ")")

if g[0] == e[0]:
	if g[1] == e[1]:
		if g[2] == e[2]:
#			print("versions are the same.")
			sys.exit(0)
		else:
			if g[2] != e[2]:
				print("Recommend using ESP-IDF " + expected + " (found " + given + ")")
				print(update)
			sys.exit(0);
	else:
#		print("minor versions differ. Please update.");
		print("*** Update required to ESP-IDF " + expected)
		print(update)
		sys.exit(0);
else:
#	print("major versions differ. Please update.");
	print("*** Update required to ESP-IDF " + expected)
	print(update)
	sys.exit(0);
```

</details>

Would you give that a try and let me know what you find? I'm particularly interested in the output of this line:

```py
print("EXPECTED ESP-IDF " + expected + " (FOUND " + given + ")")
```

Thanks for the help. (I don't have a Windows machine to try this on!)

Sineos, 2022-12-27:
Bummer, I just "fixed" my issue last night: Fixing here means deleting the entire ESP-IDF and installing v4.4.2 via the offline installer. So I have now compile my first node-red-mcu-moddable-cool-thing-stuff 🚀 

Originally I had v4.4.3 installed and added v4.4.2 with git-cloning the v4.4.2 tag and following https://docs.espressif.com/projects/esp-idf/en/v3.3/get-started-cmake/index.html#get-started-get-esp-idf-cmake
So either I have messed it up along the way (entirely possible) or the whole environment is not particularly happy when two IDFs are installed.
Strange tho since all `echo %IDF_something%` had returned the correct paths.

Thanks a lot for your support!

phoddie, 2022-12-27:
No worries. Glad you have things running.

To help diagnose this sort of thing in the future, I made changes to output the ESP-IDF version whenever it doesn't match the expected version (basically (1) above).

closed 2022-12-27 by phoddie