Moddable-OpenSource/moddable #999: [Driver Support] Sensor Wish List

Sineos, 2022-12-30:
This is truly and awesome project.
To make a real difference to its "competitors" like ESPHome, I would wish for some more sensor support:

| MFR No     | Manufacturer | Functions                  | Datasheet                                                                                                                                                                                              | Existing Implementation                                                                                    | Existing Implementation                                                                |
| ---------- | ------------ | -------------------------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ | ---------------------------------------------------------------------------------------------------------- | -------------------------------------------------------------------------------------- |
| WZ-S       | DART         | Formaldehyde               | [Link](https://www.dart-sensors.com/wp-content/uploads/2017/07/formaldehyde_module_operation-manual.pdf) | [Link](https://github.com/Gillwindy/WZ-S)                                     |                                                                                        |
| 006-1-0100 | SenseAir     | CO2                        | [Link](https://rmtplusstoragesenseair.blob.core.windows.net/docs/Dev/publicerat/PSP12930.pdf)                         | [Link](https://github.com/iotta/sunrise-rs)                                 |                                                                                        |
| 004-0-0010 | SenseAir     | CO2                        | [Link](https://rmtplusstoragesenseair.blob.core.windows.net/docs/publicerat/PSP103.pdf)                                     | [Link](https://github.com/jcomas/S8_UART)                                    | [Link](https://github.com/danielbernalb/LibreCO2) |
| PMSx003    | Plantower    | PM2.5                      | [Link](https://www.plantower.com/static/upload/file/20220627/1656292073878896.pdf)                                               | [Link](https://github.com/fu-hsi/PMS)                                             |                                                                                        |
| SPS30     | Sensirion        | PM2.5 | [Link](https://sensirion.com/media/documents/8600FF88/616542B5/Sensirion_PM_Sensors_Datasheet_SPS30.pdf)       | [Link](https://github.com/Sensirion/embedded-sps) |                                                                                        |
| BME68x     | Bosch        | VOC, Temperature, Humidity | [Link](https://www.bosch-sensortec.com/media/boschsensortec/downloads/datasheets/bst-bme680-ds001.pdf)       | [Link](https://github.com/BoschSensortec/BME68x-Sensor-API) |                                                                                        |

phoddie, 2022-12-30:
@Sineos – great idea to build out some goals. And **very helpful** to have links to both the datasheet and implementations. I hope other will contribute to this wish list.

The BME68x comes up quite often. From an implementation perspective, have two paths available. The first is to re-use the Bosch native C library (which is what Arduino does). This is a little more complex to implement but leverages the manufacturer's work. The second is to implement it entirely in JavaScript. This is likely smaller and more portable, but it will be more work.

By contrast, the WZ-S and PMSx003 example implementations are small and straightforward.

The Moddable team is committed to growing the suite of supported sensors. We can look into some of these. We are always happy to have contributions and support others working on sensor implementations.

andycarle, 2022-12-30:
I already have a BME680 on its way to me. Once that arrives I'll take a closer look at options for supporting it.

Sineos, 2023-01-12:
To continue my wish list:

| MFR No     | Manufacturer | Functions                  | Datasheet                                                                                                                                                                                              | Existing Implementation                                                                                    | Existing Implementation                                                                |
| ---------- | ------------ | -------------------------- | ------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------ | ---------------------------------------------------------------------------------------------------------- | -------------------------------------------------------------------------------------- |
| [EMC2101](https://www.adafruit.com/product/4808)       | Adafruit / Microchip | Fan Controller| [Link](https://ww1.microchip.com/downloads/aemDocuments/documents/MSLD/ProductDocuments/DataSheets/EMC2101-Data-Sheet-DS20006703.pdf) | [Link](https://github.com/adafruit/Adafruit_EMC2101)                                     |                                                                                        |

dtex, 2023-02-10:
I compiled a [wish list over at Johnny-Five](https://github.com/rwaldron/johnny-five/wiki/Requested-Features). It's divided into boards, classes, devices, and class features. Each has links to the original issues where the request came up. Sensors are found under devices.