Moddable-OpenSource/moddable #1005: SPI , device crash and reboot without message on use wrong SPI port

MKGaru, 2023-01-15:
**Build environment:** Linux
**Target device:** NodeMCU ESP32

**Description**
device crash and reboot without message on use wrong SPI port

**Steps to Reproduce**
```javascript
new device.io.SPI({
	clock: device.SPI.default.clock,
	out: device.SPI.default.out,
	hz: 40_000_000,
	active: 0,
})
```

**Expected behavior**
Appropriate error messages would be developer friendly:
`xsUnknownError("port required");`

https://github.com/Moddable-OpenSource/moddable/blob/491f8c5a83d860447f8d29208e07d59eae4d98d6/modules/io/spi/esp/spi.c#L126

phoddie, 2023-01-15:
When `port` is omitted, the call to `xsmcGet` returns `undefined`. When `undefined` is passed to `xsmcToInteger`, JavaScript type-coercion rules convert it 0. So you are using SPI port 0 (`SPI1_HOST`). There is no default SPI port on ESP32, so the constructor should throw. The lightweight fix is to replace line 127:

```c
tmp = builtinGetSignedInteger(the, &xsVar(0));
```

For `undefined`, `builtinGetSignedInteger` throws `"invalid"`. That's not the most friendly message, but it enough to know there is a problem to correct. With more code, of course, we give a more detailed message, but I'm not sure if it is worth the additional code size (I always keep in mind that flash space is limited). What do you think?

MKGaru, 2023-01-15:
I agree with the content of the proposal.
Even if the message is not strict, if an error is thrown, xsbug seems to be able to identify the problem easily.

closed 2023-02-01 by phoddie