Moddable-OpenSource/moddable #1009: xsbug: Debug status gets closed when you try to expand the tree data in left hand panels

ralphwetzel, 2023-01-20:
**Build environment:** macOS
**Target device:** m5stack Atom Lite

**Description**
When trying to examine the debug status via the left hand panels, `xsbug` tends to close the status.
See what happens when clicking on `this`.

https://user-images.githubusercontent.com/16342003/213681614-a495514a-d02c-48bf-a109-91369d8ad833.mov

This is the communication protocol between the MCU & `xsbug`:

```
[2023-01-20T11:26:24.894] [TRACE] xsbug - >> <xsbug><log path="/Users/ralph/Projekte/Moddable/moddable/modules/io/socket/lwip/tcp.c" line="390"># Exception: write: write failed!
</log></xsbug>

[2023-01-20T11:26:24.896] [TRACE] xsbug - >> <xsbug><log>XS abort: unhandled exception&#10;</log></xsbug>

[2023-01-20T11:26:24.900] [TRACE] xsbug - >> <xsbug><samples>2,0,0,0,1,34,0,0,0,0,362,1644,34,51,27604,110592,74880,106496,3696,8192,22,28,0,0,67</samples></xsbug>

[2023-01-20T11:26:24.997] [TRACE] xsbug - >> <xsbug><frames><frame name="write" value="@3FFDB0FC"/><frame name="MQTTClient.prototype.write" value="@3FFDB30C" path="/Users/ralph/Projekte/Moddable/moddable/examples/io/tcp/mqttclient/mqttclient.js" line="174"/><frame name="MQTTBrokerNode.prototype.onMessage" value="@3FFDB3FC" path="/Users/ralph/Projekte/node-red-mcu-plugin/node-red-mcu/nodered.js" line="1172"/><frame name="MQTTOutNode.prototype.onMessage" value="@3FFDB50C" path="/Users/ralph/Projekte/node-red-mcu-plugin/node-red-mcu/nodered.js" line="1268"/><frame name="Node.prototype.receive" value="@3FFDB5CC" path="/Users/ralph/Projekte/node-red-mcu-plugin/node-red-mcu/nodered.js" line="300"/><frame name="mcu.deliver" value="@3FFDB68C" path="/Users/ralph/Projekte/node-red-mcu-plugin/node-red-mcu/nodered.js" line="138"/><frame name="(host)" value="@3FFDB6FC"/></frames><local name="write" value="@3FFDB0FC"><property flags=" cew_" name="(return)" value="undefined"/><property flags=" cew_" name="new.target" value="undefined"/><property flags=" cew_" name="(function)" value="(host function)"/><property flags="+cew_" name="this" value="@3FFFC79C"/><property flags="+cew_" name="arg(0)" value="@3FFF660C"/></local><global><property flags="-cew_" name="(..)" value="@3F453790"><property flags="+cEwM" name="AggregateError" value="@3F468C70"/><property flags="+cEwM" name="Array" value="@3F461120"/><property flags="+cEwM" name="ArrayBuffer" value="@3F461360"/><property flags="+cEwM" name="Atomics" value="@3F459110"/><property flags="+cEwM" name="BigInt" value="@3F461DE0"/><property flags="+cEwM" name="BigInt64Array" value="@3F466370"/><property flags="+cEwM" name="BigUint64Array" value="@3F466230"/><property flags="+cEwM" name="Boolean" value="@3F4688B0"/><property flags="+cEwM" name="DataView" value="@3F469180"/><property flags="+cEwM" name="Date" value="@3F4635B0"/><property flags="+cEwM" name="Error" value="@3F468C40"/><property flags="+cEwM" name="EvalError" value="@3F468CD0"/><property flags="+cEwM" name="FinalizationRegistry" value="@3F469C00"/><property flags="+cEwM" name="Float32Array" value="@3F466110"/><property flags="+cEwM" name="Float64Array" value="@3F466050"/><property flags="+cEwM" name="Int16Array" value="@3F465DF0"/><property flags="+cEwM" name="Int32Array" value="@3F465B80"/><property flags="+cEwM" name="Int8Array" value="@3F465EB0"/><property flags="+cEwM" name="JSON" value="@3F462E60"/><property flags="+cEwM" name="Map" value="@3F4696C0"/><property flags="+cEwM" name="Math" value="@3F453CA0"/><property flags="+cEwM" name="ModuleSource" value="@3F469DD0"/><property flags="+cEwM" name="Number" value="@3F4582B0"/><property flags="+cEwM" name="Object" value="@3F455F00"/><property flags="+cEwM" name="Promise" value="@3F45DF30"/><property flags="+cEwM" name="Proxy" value="@3F457C20"/><property flags="+cEwM" name="RangeError" value="@3F468D10"/><property flags="+cEwM" name="ReferenceError" value="@3F468D50"/><property flags="+cEwM" name="Reflect" value="@3F458DB0"/><property flags="+cEwM" name="RegExp" value="@3F468BA0"/><property flags="+cEwM" name="Set" value="@3F4698E0"/><property flags="+cEwM" name="SharedArrayBuffer" value="@3F4692D0"/><property flags="+cEwM" name="String" value="@3F460CA0"/><property flags="+cEwM" name="Symbol" value="@3F457C50"/><property flags="+cEwM" name="SyntaxError" value="@3F468DB0"/><property flags="+cEwM" name="TypeError" value="@3F468DC0"/><property flags="+cEwM" name="TypedArray" value="@3F460410"/><property flags="+cEwM" name="URIError" value="@3F468E50"/><property flags="+cEwM" name="Uint16Array" value="@3F465800"/><property flags="+cEwM" name="Uint32Array" value="@3F4657A0"/><property flags="+cEwM" name="Uint8Array" value="@3F465A00"/><property flags="+cEwM" name="Uint8ClampedArray" value="@3F4656B0"/><property flags="+cEwM" name="WeakMap" value="@3F469920"/><property flags="+cEwM" name="WeakRef" value="@3F469BB0"/><property flags="+cEwM" name="WeakSet" value="@3F469AC0"/><property flags="+cEwM" name="decodeURI" value="@3F4658E0"/><property flags="+cEwM" name="decodeURIComponent" value="@3F470380"/><property flags="+cEwM" name="encodeURI" value="@3F46FF40"/><property flags="+cEwM" name="encodeURIComponent" value="@3F46FEF0"/><property flags="+cEwM" name="escape" value="@3F46FE00"/><property flags="+cEwM" name="isFinite" value="@3F470190"/><property flags="+cEwM" name="isNaN" value="@3F46EC40"/><property flags="+cEwM" name="parseFloat" value="@3F470050"/><property flags="+cEwM" name="parseInt" value="@3F46EBA0"/><property flags="+cEwM" name="trace" value="@3F465630"/><property flags="+cEwM" name="unescape" value="@3F46FB00"/><property flags=" CEWM" name="Infinity" value="Infinity"/><property flags=" CEWM" name="NaN" value="NaN"/><property flags=" CEWM" name="undefined" value="undefined"/><property flags="+cEwM" name="Compartment" value="@3F469D80"/><property flags="+cEwM" name="Function" value="@3F467C30"/><property flags="+cEwM" name="eval" value="@3F46FBE0"/><property flags="+cewM" name="System" value="@3F453C50"/><property flags="+cewM" name="device" value="@3F460030"/><property flags="+cewM" name="process" value="@3F469D50"/><property flags="+cewM" name="&#60;xsbug:script&#62;" value="@3F469D90"/><property flags=" cewM" name="setInterval" value="(host function)"/><property flags=" cewM" name="clearInterval" value="(host function)"/><property flags=" cewM" name="setTimeout" value="(host function)"/><property flags=" cewM" name="clearTimeout" value="(host function)"/><property flags="+cewM" name="console" value="@3F45EBB0"/><property flags="+cewM" name="Buffer" value="@3F461920"/><property flags="+cewM" name="RED" value="@3F45D510"/><property flags="+cewM" name="module" value="@3F46C630"/><property flags="+cewM" name="require" value="@3F4695D0"/></property><property flags="+cEw_" name="global" value="@3FFE4A3C"/><property flags="+cEw_" name="globalThis" value="@3FFE4A3C"/><property flags="+cew_" name="button" value="@3FFE4BCC"/><property flags="+cew_" name="lights" value="@3FFE4D4C"/><property flags="+cew_" name="flows" value="@3FFE61DC"/><property flags="+cew_" name="globalContext" value="@3FFE63CC"/><property flags=" cew_" name="[Symbol.iterator]" value="undefined"/></global><grammar><node flags="+cew_" name="main" value="@3FFE517C"/><node flags="+cew_" name="flows" value="@3FFE54AC"/><node flags="+cew_" name="node-red-contrib-semaphore" value="@3FFE557C"/><node flags="+cew_" name="node-red-node-smooth" value="@3FFE564C"/><node flags="+cew_" name="semaphore" value="@3FFE67CC"/><node flags="+cewM" name="pins/digital" value="@3F453140"/><node flags="+cewM" name="pins/digital/monitor" value="@3F453150"/><node flags="+cewM" name="m5button" value="@3F453160"/><node flags="+cewM" name="neopixel" value="@3F453170"/><node flags="+cewM" name="setup/target" value="@3F453180"/><node flags="+cewM" name="time" value="@3F453020"/><node flags="+cewM" name="timer" value="@3F453110"/><node flags="+cewM" name="Resource" value="@3F452FE0"/><node flags="+cewM" name="deepEqual" value="@3F453000"/><node flags="+cewM" name="structuredClone" value="@3F453780"/><node flags="+cewM" name="modules" value="@3F453190"/><node flags="+cewM" name="base64" value="@3F453120"/><node flags="+cewM" name="hex" value="@3F453050"/><node flags="+cewM" name="wifi" value="@3F4531A0"/><node flags="+cewM" name="socket" value="@3F452F50"/><node flags="+cewM" name="setup/network" value="@3F4531B0"/><node flags="+cewM" name="mc/config" value="@3F4531C0"/><node flags="+cewM" name="wifi/connection" value="@3F4531D0"/><node flags="+cewM" name="net" value="@3F4531E0"/><node flags="+cewM" name="sntp" value= [... cut ...] "/><node flags="+cewM" name="commands" value="@3F4530F0"/><node flags="+cewM" name="heartbeat" value="@3F4535C0"/></grammar></xsbug>

[2023-01-20T11:27:05.855] [TRACE] xsbug - << 
<toggle id="@3FFFC79C"/>

[2023-01-20T11:27:06.173] [TRACE] xsbug - >> <xsbug><login name="main" value="XS"/></xsbug>

[2023-01-20T11:27:06.195] [TRACE] xsbug - << 
<set-all-breakpoints></set-all-breakpoints>
```

I looks like the request of `xsbug` to send additional data for `@3FFFC79C` triggers a reset of the MCU...

phoddie, 2023-01-21:
> I looks like the request of xsbug to send additional data for @3FFFC79C triggers a reset of the MCU...

Agreed. That is what the trace suggests is happening.

Unfortunately, I wasn't able to reproduce the problem. Here's what I tried:

- Used M5Atom Matrix
- Simple Node-RED flow that posts random numbers to an MQTT endpoint
- Build and run with `mcconfig -d -m -p esp32/m5atom_matrix ssid="" password=""`
- Modified lwip/tcp.c to throw an exception on the 20th `write` at the same point as the exception you show

I captured the xsbug traffic using Wireshark on `localhost`. Here's [the log](https://github.com/Moddable-OpenSource/moddable/files/10471532/xsbug.txt),. Perhaps it will offer some clue?

ralphwetzel, 2023-01-21:
> I looks like the request of `xsbug` to send additional data for `@3FFFC79C` triggers a reset of the MCU...

> I captured the xsbug traffic using Wireshark

The data package the MCU collects & sends back to xsbug is almost 10kb heavy.
Assumption: While processing this data the MCU internally hits an exception (e.g. `memory full`) & resets.

phoddie, 2023-01-22:
The MCU streams the information it sends to xsbug. It is never entirely in memory at one time. It can be arbitrarily large and not fail because of limited memory. xsbug was designed for resource-constrained environments. ;)