Moddable-OpenSource/moddable #1011: ARM-based Mac: mcsim/mc.so incompatible architecture

rei-vilo, 2023-01-22:
Targeting the simulator raises an error on an ARM-based MacBook.

> tech.moddable.mcsim/mc.so' (mach-o file, but is an incompatible architecture (have '×86_64', need 'arm64')),

[image]

Environment
+ `git describe --abbrev=7 --always --long`gives 3.6.0-55-g5569674
+ MacBook Air M1 2020
+ macOS 13.1 (22C65)

phoddie, 2023-01-22:
The Moddable SDK has built on Apple Silicon since shortly after its launch. I develop on an M2.

It looks like you may have an ARM build of the simulator and an x86 build of the project (`mc.so`). If you installed using xs-dev it would install ARM version of the tools binaries.

Did you migrate to an Apple Silicon machine recently? If so, you may need to update Xcode and the its Command Line tools to get the ARM versions.

rei-vilo, 2023-01-23:
Performing a clean install with `xs-dev` solved the issue.

The previous install was performed according to the [Updating procedure](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#mac-update).

closed 2023-01-23 by rei-vilo