Moddable-OpenSource/moddable #1012: esp32/saola_wroom & esp32/saola_wrover fail to build

ralphwetzel, 2023-01-24:
Relaying a [report from the Node-RED Forum](https://discourse.nodered.org/t/node-red-mcu-plugin-v1-0-integrating-node-red-mcu-edition-into-the-node-red-editor/72693/21?u=ralphwetzel): I can confirm that a build fails on both devices.
This does **not** seem to be related though to Node-RED MCU Edition...

phoddie, 2023-01-24:
Looks like the pulse width input is different on the S2 -- or maybe just not present. 

```
/Users/hoddie/Projects/moddable/modules/io/pulsewidth/esp32/_pulsewidth.c:58:25: error: unknown type name 'mcpwm_unit_t'
 static bool pw_callback(mcpwm_unit_t mcpwm, mcpwm_capture_channel_id_t cap_sig, const cap_event_data_t *edata, void *arg);
                         ^~~~~~~~~~~~
```

@andycarle, would you take a look?

@ralphwetzel - the short-term solution is to just to have the developer remove the pulsewidth module from their local copy of `manifest_runtime`. It is not required and is rarely used at the moment (only by the HC-SR04 proximity sensor).

jmorris644, 2023-01-24:
I am the developer in question, where can I find the manifest_runtime?

ralphwetzel, 2023-01-24:
@jmorris644 
In your Node-RED `<user-dir>` ... which should be `~/.node-red`  ... and then `node_modules/@ralphwetzel/node-red-mcu-plugin/node-red-mcu/manifest_runtime.json`.

andycarle, 2023-01-24:
The issue is, indeed, that Pulse Width input is simply not available on the ESP32-S2. I've pushed a change locally to disable it. That will be live in our next GitHub push.

jmorris644, 2023-01-25:
> 

Sorry but I do not see a module in my manifest_runtime.json that looks like the pulsewidth

phoddie, 2023-01-25:
Apologies for the misleading suggestion. I have recently been working on pulsewidth support and got this build error mixed up with that.

The patch @andycarle developed will be live by the end of the week. Until then, you can apply it manually to the Moddable SDK's [ESP32 pulsewidth implementation](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/pulsewidth/esp32/_pulsewidth.c).

<details>
<summary>$MODDABLE/modules/io/pulsewidth/esp32/_pulsewidth.c</summary>

```c
/*
 * Copyright (c) 2022-2023 Moddable Tech, Inc.
 *
 *   This file is part of the Moddable SDK Runtime.
 * 
 *   The Moddable SDK Runtime is free software: you can redistribute it and/or modify
 *   it under the terms of the GNU Lesser General Public License as published by
 *   the Free Software Foundation, either version 3 of the License, or
 *   (at your option) any later version.
 * 
 *   The Moddable SDK Runtime is distributed in the hope that it will be useful,
 *   but WITHOUT ANY WARRANTY; without even the implied warranty of
 *   MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *   GNU Lesser General Public License for more details.
 * 
 *   You should have received a copy of the GNU Lesser General Public License
 *   along with the Moddable SDK Runtime.  If not, see <http://www.gnu.org/licenses/>.
 *
 */

#include "xsmc.h"			// xs bindings for microcontroller
#include "xsHost.h"			// esp platform support
#include "mc.xs.h"			// for xsID_* values

#if defined(SOC_MCPWM_SUPPORTED)
#include "builtinCommon.h"

#include "driver/mcpwm.h"

enum {
    kPulseWidthRisingToFalling = 1,
    kPulseWidthFallingToRising = 2,
    kPulseWidthRisingToRising = 3,
    kPulseWidthFallingToFalling = 4
};

enum {
    kPulseWidthFloating = 0,
    kPulseWidthPullUp = 1,
    kPulseWidthPullDown = 2
};

struct PulseWidthRecord {
	uint32_t	pin;
	xsSlot		obj;
	uint32_t    value;
    uint8_t     channel;
    uint8_t     edges;
    uint8_t     hasOnReadable;

    // Allocated only if onReadable callback present
	xsMachine	*the;
	xsSlot		*onReadable;
};
typedef struct PulseWidthRecord PulseWidthRecord;
typedef struct PulseWidthRecord *PulseWidth;

static bool pw_callback(mcpwm_unit_t mcpwm, mcpwm_capture_channel_id_t cap_sig, const cap_event_data_t *edata, void *arg);
static void pulseWidthDeliver(void *the, void *refcon, uint8_t *message, uint16_t messageLength);
void xs_pulsewidth_destructor(void *data);
void xs_pulsewidth_mark(xsMachine* the, void *it, xsMarkRoot markRoot);
static uint8_t mcpwmFromChannel(uint8_t channel, mcpwm_unit_t *unit, mcpwm_io_signals_t *signal, mcpwm_capture_signal_t *select);
[... 248 lines cut ...]
        *signal = MCPWM_CAP_1;
        *select = MCPWM_SELECT_CAP1;
        result = 0b10;
    } else if (x & 0b100) {
        *signal = MCPWM_CAP_2;
        *select = MCPWM_SELECT_CAP2;
        result = 0b100;
    }

    if (*unit == MCPWM_UNIT_1)
        result <<= 3;

    return result;
}

#else // ! SOC_MCPWM_SUPPORTED
void xs_pulsewidth_read(xsMachine *the) {}
void xs_pulsewidth_mark(xsMachine* the, void *it, xsMarkRoot markRoot) {}
void xs_pulsewidth_destructor(void *data) {}
void xs_pulsewidth_constructor(xsMachine *the) {}
void xs_pulsewidth_close(xsMachine *the) {}
#endif
```

</details>

jmorris644, 2023-10-13:
As I am getting back into more testing, I did a clean install on my server. I still got this error and had to re-paste the above solution. Maybe this commit did not get into the recent release?

phoddie, 2023-10-13:
Welcome back!

I just did a quick test, building a Node-RED flow, which includes the pulse width input module. I build that for both esp32/saola_wroom and esp32/wrover targets. They both succeeded. I confirmed it is building esp32/_pulsewidth.c and that on the ESP32-S2 target it uses the stub case added above (`#else // ! SOC_MCPWM_SUPPORTED`. 

Not sure what is going on. This [is the relevant code](https://github.com/Moddable-OpenSource/moddable/blob/6613fc5d8e9556fb6255d2c8379acaaab36dd387/modules/io/pulsewidth/esp32/_pulsewidth.c#L290) in the current Moddable SDK repository. Is it possible you are somehow still using older sources?

jmorris644, 2023-10-13:
Ok, I will make sure I do a total reinstall rather than an upgrade. I will report back. Thanks!

jmorris644, 2023-10-15:
I deleted the esp32 and Projects dirs. Did a complete reinstall. It works now. I must have done just an update previously. My apologies for bothering you.

phoddie, 2023-10-15:
No problem. Glad it is working for you now.