Moddable-OpenSource/moddable #1016: xsbug protocol does not start up reliably on Linux

tve, 2023-01-29:
**Build environment:** Linux, Arch
**Target device:** NodeMCU board w/esp32

**Description**
When serial2xsbug starts up it frequently hangs after set-all-breakpoints.

**Steps to Reproduce**
1. Using the chapter 1 host from the IoT book (`iot-product-dev-book/ch1-gettingstarted/host`) build and load the host using `mcconfig -d -m -p esp32/nodemcu`
2. Observe xsbug starting and (hopefully) the expected "Device flashed..." log line.
3. In serial2xsbug.h enable mxTrace and mxTraceCommands
5. Run just serial2xsbug several times, for me it hangs 2-3 times out of 10, sometimes more consistently, adjust the commandline for your port: `BUG_PORT=5002 XSBUG_HOST=localhost serial2xsbug /dev/ttyUSB0 460800 8N1` (you can also re-run mcconfig, it just takes longer...)
6. Output when serial2xsbug hangs:
```
> SBUG_PORT=5002 XSBUG_HOST=localhost serial2xsbug /dev/ttyUSB0 460800 8N1
### fxRestart
xxxxxxxxxxxxxxxxxxxxxx<?xs-00000000?>

<?xs.3FFB4264?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB4264?>
<?xs.3FFB4264?>
<set-all-breakpoints></set-all-breakpoints>

<?xs-3FFB4264?>
```
(There is about a 1 second delay between the `<set-all-breakpoints...` line and the very last line.)

**Expected behavior**
A successful start-up:
```
> SBUG_PORT=5002 XSBUG_HOST=localhost serial2xsbug /de
v/ttyUSB0 460800 8N1                                                          
### fxRestart                                                                 
xxxxxxxxxxxxxxxxxxxxxxx<?xs-00000000?>                                        
                                                                              
<?xs.3FFB4264?>                                                               
<xsbug><login name="main" value="XS"/></xsbug>                                
### set time                                                                  
                                                                              
<?xs#3FFB4264?>                                                               
<?xs.3FFB4264?>                                                               
<set-all-breakpoints></set-all-breakpoints>                                   
                                                                              
<?xs.3FFB4264?>                                                               
<xsbug><instruments><instrument name="Pixels drawn" value=" pixels"/><instrume
nt name="Frames drawn" value=" frames"/><instrument name="Network bytes read" 
value=" bytes"/><instrument name="Network bytes written" value=" bytes"/><inst
rument name="Network sockets" value=" sockets"/><instrument name="Timers" valu
e=" timers"/><instrument name="Files" value=" files"/><instrument name="Poco d
isplay list used" value=" bytes"/><instrument name="Piu command List used" val
ue=" bytes"/><instrument name="SPI flash erases" value=" sectors"/><instrument
 name="Turns" value=" turns"/><instrument name="System bytes free" value=" byt
es"/><instrument name="CPU 0" value=" percent"/><instrument name="CPU 1" value
=" percent"/><instrument name="Chunk used" value=" / "/><instrument name="Chun
k available" value=" bytes"/><instrument name="Slot used" value=" / "/><instru
ment name="Slot available" value=" bytes"/><instrument name="Stack used" value
=" / "/><instrument name="Stack available" value=" bytes"/><instrument name="G
arbage collections" value=" times"/><instrument name="Keys used" value=" keys"
/><instrument name="Modules loaded" value=" modules"/><instrument name="Parser
 used" value=" bytes"/><instrument name="Floating Point" value=" operations"/>
</instruments></xsbug>

<?xs.3FFB4264?>
<xsbug><log>Device flashed. Ready to install apps.&#10;</log></xsbug>
```

phoddie, 2023-01-29:
Thank you for the report.

> I'm concluding that it's the code in the MCU that crashes

That is contradicted by the `<?xs-3FFB4264?>` from the MCU in the trace.

phoddie, 2023-01-29:
It looks more like the MCU didn't receive the command from the Linux device. It is timing out waiting for the initial set-up to complete. An unreliable serial connection would do that. You could try stepping back to a slower baud rate Try setting 

```
DEBUGGER_SPEED=115200 mcconfig...
```

tve, 2023-01-29:
I see, so you assume that the last `<?xs-3FFB4264?>` is a retry because the `<set-all-breakpoints></set-all-breakpoints>` wasn't properly received.

I did some more troubleshooting in the meantime and when it hangs, in `fxDescribeInstrumentation` https://github.com/Moddable-OpenSource/moddable/blob/public/xs/sources/xsDebug.c#L2339 `fxIsConnected()` returns false, that's why `<xsbug><instruments><instrument ...` isn't being sent by the MCU.

I will try a lower serial rate. I had tried a completely different board with a (genuine) FTDI adapter (whereas the NodeMCU uses a cp210x). Thanks for the suggestions!

phoddie, 2023-01-29:
> so you assume that the last <?xs-3FFB4264?> is a retry

It is not a retry. That is a [disconnect](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/xs/platforms/esp/xsPlatform.c#L488). 

If the MCU does not receive a valid command on the xsbug connection [within 2 seconds](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/xs/platforms/esp/xsPlatform.c#L685-L688) of sending the `<login` it concludes no debugger is available and continues without it.  The disconnect message indicates that no valid command was received.

tve, 2023-01-29:
Ah, thanks for the clarification! (If there's a doc I should read I'd love to.)
I had written down that the `-` after `xs` results in closing the network but then overlooked it.
That explains why `the->connection` is falsy by the time fxDescribeInstrumentation comes around. OK, so I need to figure out what the MCU is expecting in response to the login and why it's not receiving that...

phoddie, 2023-01-29:
> ...I need to figure out what the MCU is expecting in response to the login and why it's not receiving that...
It is expecting any valid command. Such as `<set-all-breakpoints></set-all-breakpoints>`. Perhaps your computer is taking longer than 2 seconds to send that?

tve, 2023-01-29:
> Perhaps your computer is taking longer than 2 seconds to send that?

It happens instantaneously.

tve, 2023-01-30:
The plot thickens... I added a buffer to capture what [fxReceiveLoop is reading](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/xs/platforms/esp/xsPlatform.c#L758). I'm printing it when fxReceive exits surrounded by {..} and with control characters shown as ~hex. Here's a successful start-up:

```
entry 0x400805d0                                                    
<?xs-00000000?>                                                     
                                                                    
<?xs.3FFB436C?>                                                     
<xsbug><login name="main" value="XS"/></xsbug>                      
### set time                                                        
                                                                    
<?xs#3FFB436C?>                                                     
<?xs.3FFB436C?>                                                     
<set-all-breakpoints></set-all-breakpoints>                         
{~0D~0A<?xs#3FFB436C?>~00~0F~09~00~00c~D72~18~FF~FF~8F~80~00~00~00~
00~0D~0A<?xs.3FFB436C?>~0D~0A<set-all-breakpoints></set-all-breakpoi
nts>~0D~0A}                                                         
<?xs.3FFB436C?>                                                        
<xsbug><log>DescribeInstrumentation&#10;</log></xsbug>              
                                                                    
<?xs.3FFB436C?>                                                     
<xsbug><log>Connected&#10;</log></xsbug>                            
                                                                    
<?xs.3FFB436C?>                                                     
<xsbug><instruments><instrument name="Pixels drawn" value
=" pixels"/><instrument name="Frames drawn" value=" frame
s"/><instrument name="Network bytes read" value=" bytes"/
...
```
Here's a failed start-up:
```
entry 0x400805d0
<?xs-00000000?>

<?xs.3FFB436C?>
<xsbug><login name="main" value="XS"/></xsbug>
### set time

<?xs#3FFB436C?>
<?xs.3FFB436C?>
<set-all-breakpoints></set-all-breakpoints>

<?xs-3FFB436C?>
{~0A<?XS#3FFB436C?>~00~0F ~00~00C~D73I~DF~DF~8F~80~00~00~00~00~0A~0A<?XS.3FFB436C?>~0A~0A<SET-ALL-BREAKPOINTS></SET-ALL-BREAKPOINTS>~0A~0A}DescribeInstrumentationNotConnected^C⏎              
```
... nope, those CAPS are not my editing...

phoddie, 2023-01-30:
That's really strange. Is it possible there is some configuration on the Linux serial driver you are using that is performing case conversion? The code in serial2xsbug for that is [here](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/tools/serial2xsbug/serial2xsbug_lin.c#L124-L166).

> BTW, what does the "fx" prefix on all the function in serial2xsbug stand for?

Function. XS coding style.

tve, 2023-01-30:
Yeah, I'm trying to understand fxOpenSerial... The first thing that stood out is that everywhere I read they start with tcgetattr to init the termios struct before modifying and setting. https://stackoverflow.com/a/6947758/3807231 even claims you get erratic behavior if you don't. I don't know I believe that, but I am observing erratic behavior...

Unfortunately, just starting with tcgetattr doesn't work 'cause some flags that need to be cleared aren't.

I'm trying to ascertain that the code wants a "completely raw" connection and will then try some sample code from the interwebs... (I don't have linux serial programming experience other than hacking around when it doesn't work.)

phoddie, 2023-01-30:
> . (I don't have linux serial programming experience other than hacking around when it doesn't work.)

That's where most everyone seems to be. It is a black art. Good luck. Looking forward to what you discover.

tve, 2023-01-30:
OK, erratic behavior explained... Looking at [fxOpenSerial](https://github.com/Moddable-OpenSource/moddable/blob/819c387912fe0d8b53582b27c5739f1c261c3688/tools/serial2xsbug/serial2xsbug_lin.c#L124-L166):
- `struct termios term` is declared at the top but not initialized
- `term.c_oflag &= ~ONLCR;` leaves all output processing flags except for ONLCR in a random state depending on the uninitialized `term` field

In addition, there are several fields that set the speed and the recommended portable method is to use `cfsetspeed()` (see the man page).

Also, there is a section in the second half of the function to set a non-standard speed (the `if (speed == BOTHER)` section) but:
- `speed` is never set to BOTHER
- this section loads a `tio2` struct and then uses that struct to update the speed, but the assignments in-between manipulate `term`, so that section is completely ineffective

Right now it seems to work reliably for me. I'm leaning towards a PR that makes three changes:
- init `term` using `cfmakeraw()` (alternatively use memset)
- set `c_oflag` fully
- set the speed using `cfsetspeed`
- fix the "other speed" section (although I don't really know how I'd test it)

Thoughts?

phoddie, 2023-01-30:
Thanks for the analysis and PR. It more-or-less makes sense (given a limited understanding of every detail). We'll review.

commit 633d9fb8 (2023-02-02): serial2xsbug_lin: fix serial port init to avoid non-determinism, fixes #1016
closed 2023-02-01 by phoddie