Moddable-OpenSource/moddable #1019: xsbug: Scrolling the code pane with a scroll wheel is unusable (Linux)

tve, 2023-01-30:
**Build environment:** Arch Linux, Wayland

**Description**
Scrolling the code pane using a mouse scroll wheel is practically unusable:
- a full finger length rotation of the scroll wheel moves the code pane by only about half a line
- the direction of scrolling is the opposite of all other apps

**Steps to Reproduce**
1. Open any long file in xsbug and try to scroll

**Other information**
I do not know whether this is related to Linux in general or to wayland or to something else.

phoddie, 2023-01-31:
Piu provides scroll wheel support for macOS, Windows, and Linux. The Linux code for that is [fairly small](https://github.com/Moddable-OpenSource/moddable/blob/4f8b09ea36419d0c7137df09ea92f9861568cc28/modules/piu/PC/lin/piuView.c#L192-L215). I suppose the challenge is finding a behavior there which works well enough across applications.

tve, 2023-01-31:
GTK... that totally leaves my domain of expertise... isn't there some metadata that GTK provides that has info about the scrolling device and direction?

phoddie, 2023-01-31:
> GTK... that totally leaves my domain of expertise... isn't there some metadata that GTK provides that has info about the scrolling device and direction?

GTK... that totally leaves my domain of expertise.