Moddable-OpenSource/moddable #1024: mcrun should have the same env vars by default

tve, 2023-02-03:
By default, manifests processed by mcconfig can use `$(MODULES)` while manifests processed by mcrun cannot. Please make the default env vars the same (I didn't check `$(PIU)`, etc). I realize this can be worked around by specifying an explicit `build` property but given that the docs for modules all have a section like
```
"include": [
		"$(MODULES)/data/base64/manifest.json"
	]
```
it's kind'a novice unfriendly.

Running mcrun without the work-around results in
```
### Error: 'undefined/data/base64/manifest.json': manifest not found!   
```

andycarle, 2023-02-04:
I agree with your bottom line point here, but your statement of the problem is a little off in a way that matters when considering solutions.

> By default, manifests processed by mcconfig can use $(MODULES) while manifests processed by mcrun cannot.

That is not quite right. What's actually happening is that most Moddable SDK examples (and, subsequently, most apps that people build) include `"$(MODDABLE)/examples/manifest_base.json"`. [`manifest_base.json`](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_base.json#L2) defines the variables that you want:

```json
	"build": {
		"BUILD": "$(MODDABLE)/build",
		"MODULES": "$(MODDABLE)/modules",
		"COMMODETTO": "$(MODULES)/commodetto",
		"PIU": "$(MODULES)/piu"
	}
```

The right solution is probably to also include that block of variables in [manifest_mod.json](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/manifest_mod.json), which most Mods include. Could you try that out and see if it fixes this issue in a way that feels right to you? Just replace the contents of `manifest_mod.json` with:

```json
{
	"build": {
		"BUILD": "$(MODDABLE)/build",
		"MODULES": "$(MODDABLE)/modules",
		"COMMODETTO": "$(MODULES)/commodetto",
		"PIU": "$(MODULES)/piu"
	},
	"platforms": {
        "esp32/m5atom_echo": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        },
        "esp32/m5atom_lite": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        },
        "esp32/m5atom_matrix": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        },
        "esp32/m5atom_u": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        },
		"esp32/m5stack_core2": {
			"build": {
				"UPLOAD_SPEED": "1500000",
				"DEBUGGER_SPEED": "921600"
			}
        },
        "esp32/m5stick_c": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        },
        "esp32/m5stick_cplus": {
            "build": {
                "UPLOAD_SPEED": "1500000",
                "DEBUGGER_SPEED": "1500000"
            }
        }
	}
}
```

tve, 2023-02-04:
Ah, thanks for the clarification, I had not looked where the variable definitions come from and had assumed they were in the tools. 🙄
I don't see why your proposed fix wouldn't work. I can't really test it on a real project 'cause ... well... pulling the base64 manifest into a mod ain't gonna work no matter what the env vars are 'cause it uses native code, ooops! 😬

phoddie, 2023-02-09:
I don't think there's a strong reason to define PIU or COMMODETTO. The modules there aren't likely to be included by a mod, but provided by the host. I'm not sure BUILD is all that useful either. There's no harm in adding them, I suppose. But MODULE makes good sense

With tomorrow's Moddable SDK update, we'll reject attempts to use a native module in a mod at build time, so mistakes will be caught early.

phoddie, 2023-05-31:
In fact, there is at least one Piu module that is JavaScript-only. So... in the interest of consistency, I've updated the build section of manifest_mod to match manifest_base.

phoddie, 2023-06-27:
Closing as proposed change committed. 

Note that there were also changes to provide much better error messages when attempting to use native modules with mcrun:

```
Native code detected:
  ~/moddable/modules/base/time/esp/modTime.c
  ~/moddable/modules/base/timer/modTimer.c
Mods cannot contain native code. Did you intend to build using mcconfig?
### Error: mod cannot contain native code
```

closed 2023-06-27 by phoddie