Moddable-OpenSource/moddable #1025: missing module can lead to system reset

tve, 2023-02-03:
**Build environment:** Arch Linux
**Target device:** NodeMCU ESP32

**Description**
Using a mod causes the system to crash. The exact circumstances are hard to pin down.
I stumbled into this when I tried to use the Base64 module in a mod not realizing that it has
native code. But I could not produce a straight-forward repro from that problem statement.
However, I could repro
the problem by stripping down my original host and mod, so I'm probably not really understanding
the exact circumstances that cause the crash.

**Steps to Reproduce**
```
> cd repro-host
> esptool.py --port /dev/ttyUSB0 erase_flash
> DEBUGGER_SPEED=115200 mcconfig -d -m -p esp32/nodemcu
```
Observe:
```
===== ReproHost starting =====
Host ready, no module installed...
Alive...
No Wi-Fi SSID
```
```
> cd repro-mod
> DEBUGGER_SPEED=115200 mcrun -d -m -p esp32/nodemcu -v
```
Observe the system crashing and restarting in xsbug:
```
===== ReproHost starting =====
Check returned
No Wi-Fi SSID

===== ReproHost starting =====
Check returned
No Wi-Fi SSID
```
And ditto in the serial2xsbug terminal:
```
### set time
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:344
load:0x40078000,len:13488
load:0x40080400,len:3520
entry 0x400805d0
### set time
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:344
load:0x40078000,len:13488
load:0x40080400,len:3520
entry 0x400805d0
```
(The `### set time` comes from my version having mxTraceCommands=1.)

**Expected behavior**
An orderly exception or error that the mod cannot be loaded due to native code.

repro-host main.js:
```
import Timer from "timer"

Timer.repeat(() => trace("Alive...\n"), 10000)

import Modules from "modules"

function launch_app() {
  if (Modules.has("check")) {
    let check = Modules.importNow("check")
    check()
    trace("Check returned\n")
    Timer.delay(2000)
    if (!Modules.has("app")) throw new Error("Module 'app' not found")
    let app = Modules.importNow("app")
    trace("===== Launching app =====\n")
    app()
    trace("App is done\n")
  } else {
    trace("Host ready, no module installed...\n")
  }
}

export default function () {
  trace("\n\n===== ReproHost starting =====\n")
  Timer.delay(1000)

  launch_app()
}
```

repro-host manifest.json:
```
{
  "include": [
    "$(MODDABLE)/examples/manifest_base.json",
    "$(MODDABLE)/examples/manifest_net.json",
    "$(MODULES)/data/text/encoder/manifest.json"
  ],
  "modules": {
    "*": [
      "./main",
      "$(MODULES)/base/modules/*",
      "$(MODULES)/network/sntp/*",
      "$(MODULES)/network/dns/moddnsparser"
    ]
  },
  "defines": {
    "XS_MODS": 1
  },
  "preload": ["modules"],
  "strip": [],
  "build": {
    "name": "ReproHost"
  },
  "creation": {
    "static": 65536,
    "keys": {
      "available": 256
    }
  }
}
```

repro-mod app.js:
```
import Base64 from "base64"

trace("Hello app module!\n")
trace(Base64.encode("Hello app module!"))
trace("\n")

export default function () {
  trace("Hello app main!\n")
  trace(Base64.encode("Hello app main!"))
  trace("\n")
}
```

repro-mod manifest.json:
```
{
  "include": [
    "$(MODDABLE)/examples/manifest_mod.json",
    "$(MODDABLE)/modules/data/base64/manifest.json"
  ],
  "modules": {
    "*": ["./*.js"]
  }
}
```

I hope this reproduces cleanly for you, I'm finding it "finnicky", i.e., depending on what I change I get the expected behavior.

phoddie, 2023-02-05:
Thanks for the example. The problem reliably reproduces using the simulator on macOS which makes debugging very easy.

The problem is the inclusion of base64. That generates byte code to initialize the native functions, which crashes at runtime because the native functions aren't present. The base64 module should just fail to compile. That way you would get an early error that it won't work. We will update mcrun to do that.

phoddie, 2023-04-26:
This has been fixed by having mcrun report an error if native function syntax is detected.

closed 2023-04-26 by phoddie