Moddable-OpenSource/moddable #1031: Fail to build with Pico SDK 1.5.0

stc1988, 2023-02-11:
**Moddable version:** 3.7.0
**Build environment:** macOS
**Target device:** pico

**Steps to Reproduce**
1. [Setup pico environment](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/pico.md#macos-setup).
2. `cd cd $MODDABLE/examples/helloworl`
3. `mcconfig -d -m -p pico`

```
make: *** No rule to make target `/Users/satoshi/.local/share/pico/pico-sdk/src/rp2_common/cyw43_driver/cyw43_bus_pio_spi.pio', needed by `/Users/Satoshi/Projects/moddable/build/tmp/pico/debug/helloworld/cyw43_bus_pio_spi.pio.h'.  Stop.
```

**Other information**
In current master branch(1.5.0),  [src/rp2_common/cyw43_driver](https://github.com/raspberrypi/pico-sdk/tree/1.5.0/src/rp2_common) is not exists.

stc1988, 2023-03-28:
Would it be better to fix below setup instructions  on [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/pico.md) ?

from 
```
git clone -b master https://github.com/raspberrypi/pico-sdk
git clone -b master https://github.com/raspberrypi/pico-examples
```

to
```
git clone -b 1.4.0 https://github.com/raspberrypi/pico-sdk
git clone -b sdk-1.4.0 https://github.com/raspberrypi/pico-examples
```

phoddie, 2023-03-28:
We would prefer to use the latest SDK. We have been working on support for that it is getting close.

Thank you for posting the temporary instructions here for developers who are blocked during this transition.

closed 2023-03-29 by mkellner