Moddable-OpenSource/moddable #1034: Fail to build with wasm

meganetaaan, 2023-02-14:
**Build environment:** Linux (Ubuntu 20.04) and moddable v3.7.0
**Target device:** wasm

**Description**

The wasm build fails. There are two problems.

  1. Tool can't build (fixed as below)
  2. When building with `-p wasm`, I get an error about timer.

**Steps to Reproduce**

- setup wasm toolkit according to [readme](https://github.com/Moddable-OpenSource/moddable/blob/886694700ec392b3c6f92cfa92cf47d8ff0b178f/documentation/devices/wasm.md)
- run step.7 of the doc.

```
cd ${MODDABLE}/build/makefiles/wasm
make
```
- Got the error below

```
# tools debug : xsc commodettoBitmap.js
make[1]: /home/sskw/.local/share/moddable/build/bin/lin/debug/xsc: コマンドが見つかりませんでした
make[1]: *** [tools.mk:296: /home/sskw/.local/share/moddable/build/tmp/wasm/debug/tools/modules/commodetto/Bitmap.xsb] エラー 127
make[1]: ディレクトリ '/home/sskw/.local/share/moddable/build/makefiles/wasm' から出ます
make: *** [makefile:29: debug] エラー 2
```

After fixing [the failed command](https://github.com/Moddable-OpenSource/moddable/blob/public/build/makefiles/wasm/tools.mk#L300) of makefile according to other lines `$(BIN_DIR)/xsc -> $(XSC)` the compile error dessapeared.

But when I try to build examples another error occurs.

<details>
<summary>build log</summary>
</pre>
sskw@sskw-XF:~/.local/share/moddable/examples/piu/balls$ mcconfig -m -d -p wasm
# xsc screen.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc main.xsb
# xsc commodetto/Bitmap.xsb
# xsc commodetto/Poco.xsb
# xsc commodetto/parseBMP.xsb
# xsc commodetto/parseRLE.xsb
# xsc piu/All.xsb
# xsc piu/Timeline.xsb
# xsc piu/CombTransition.xsb
# xsc piu/MC.xsb
# xsc piu/WipeTransition.xsb
# xsc mc/config.xsb
# xsid screen.c.xsi
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
# xsid timer.c.xsi
# xsid Resource.c.xsi
# xsid modInstrumentation.c.xsi
# xsid piuDie.c.xsi
# xsid piuFont.c.xsi
# xsid piuImage.c.xsi
# xsid piuRegion.c.xsi
# xsid piuTexture.c.xsi
# xsid piuView.c.xsi
# xsid commodettoBitmap.c.xsi
# xsid commodettoPoco.c.xsi
# xsid commodettoPocoBlit.c.xsi
# xsid cfeBMF.c.xsi
# xsid commodettoParseBMP.c.xsi
# xsid commodettoParseRLE.c.xsi
[... 77 lines cut ...]
# cc piuTransition.c.o
# cc mc.js
wasm-ld: error: duplicate symbol: xs_timer_set
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_repeat
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_schedule
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_clear
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_delay
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
>>> defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o
emcc: error: '/home/sskw/.local/share/wasm/emsdk/upstream/bin/wasm-ld -o /home/sskw/.local/share/moddable/build/bin/wasm/debug/balls/mc.wasm /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/wasm_xs.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsAll.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsAPI.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsArguments.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsArray.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsAtomics.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsBigInt.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsBoolean.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsCode.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsCommon.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsDataView.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsDate.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsDebug.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsError.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsFunction.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsGenerator.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsGlobal.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsJSON.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsLexical.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsMapSet.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsMarshall.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsMath.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsMemory.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsModule.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsNumber.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsObject.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsPlatforms.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsPromise.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsProperty.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsProxy.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsRegExp.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsRun.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsScope.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsScript.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsSourceMap.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsString.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsSymbol.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsSyntaxical.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsTree.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsType.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsdtoa.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsmc.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/lib/xsre.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/mc.xs.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/mc.resources.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/screen.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTime.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/Resource.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modInstrumentation.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuDie.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuFont.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuImage.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuRegion.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuTexture.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuView.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/commodettoBitmap.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/commodettoPoco.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/commodettoPocoBlit.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/cfeBMF.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/commodettoParseBMP.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/commodettoParseRLE.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuApplication.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuBehavior.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuColor.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuColumn.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuContainer.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuContent.c.o /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/piuLabel.c.o /home/ss [... cut ...] tore --export=stackAlloc --export=__wasm_call_ctors --export=__errno_location --export=__get_temp_ret --export=__set_temp_ret --export=malloc --export=free --export=saveSetjmp --export=setThrew --export-table -z stack-size=65536 --initial-memory=16777216 --no-entry --max-memory=2147483648 --stack-first' failed (returned 1)
make: *** [/home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/makefile:341: /home/sskw/.local/share/moddable/build/bin/wasm/debug/balls/mc.js] エラー 1
</pre>
</details>

meganetaaan, 2023-03-11:
The post above was ambiguous. The "another error" is this part.

```
wasm-ld: error: duplicate symbol: xs_timer_repeat

defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_schedule

defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_clear

defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o

wasm-ld: error: duplicate symbol: xs_timer_delay

defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/modTimer.c.o
defined in /home/sskw/.local/share/moddable/build/tmp/wasm/debug/balls/timer.c.o
emcc: error: '/home/sskw/.local/share/wasm/emsdk/upstream/bin/wasm-ld -o /home/sskw/.local/share/moddable/build/bin/wasm/debug/balls/mc.wasm
...
```

I checked the [current implementations of timer module](https://github.com/Moddable-OpenSource/moddable/tree/public/modules/base/timer/wasm). Other platforms [follow the latest interface of the Timer class](https://github.com/Moddable-OpenSource/moddable/commit/961d926f85893b0e45fe95ce30fa31c3eeebaa1b), but it seems that the wasm timer implementation does not follow it?

phoddie, 2023-03-14:
Thanks for the additional details. It looks like something got out of sync in the building of timer for wasm. While other platforms use both modTimer.c and a platform specific timer.c, the wasm port merges those into its timer.c. That's not incorrect, but the inconsistency seems to have created a build mixup. I think [this is the commit](https://github.com/Moddable-OpenSource/moddable/commit/2656c45148335b06f049e4edcaad6552b012bc75) that introduced the problem, specifically with simulators/manifest.json. Unfortunately, my wasm tooling is not set-up, so I can't easily try. Maybe you can?

meganetaaan, 2023-03-27:
After reverting [the commit](https://github.com/Moddable-OpenSource/moddable/commit/2656c45148335b06f049e4edcaad6552b012bc75) Moddable successfully builds the wasm target!

https://twitter.com/meganetaaan/status/1640393426243035136

I believe our next step should be to make `build/simulators/manifest.json` compatible with both 'wasm' and other simulator platforms.

phoddie, 2023-03-27:
> After reverting [the commit](https://github.com/Moddable-OpenSource/moddable/commit/2656c45148335b06f049e4edcaad6552b012bc75) Moddable successfully builds the wasm target!

That makes sense. Thank you for confirming.