Moddable-OpenSource/moddable #1038: Build target xsbug fails

bartbutenaers, 2023-02-18:
**Build environment:** Raspberry Pi 4 (64 bit Raspberry OS)
**Target device:** WT32-ETH01

**Description**
A few weeks ago @andycarle has supported wt32-eth01 boards for me.  First I had some troubles to get the Node-RED sidebar working, but @ralphwetzel has helped me through that with a lot of patience.  But now I am stuck in the build target xsbug phase.

**Steps to Reproduce**
1. I put the WT32-eth01 device in programming mode, by connecting pin IO0 to GND.
2. In the sidebar I choose build target ***"build"***, which seem to be successfull:
   ```
   Starting build process...
   Host system check: #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023
   MCU Build system check: p1.1.2-post + #8224a11 @ m3.6.0-104-g819c387
   HOME directory check: /home/pi
   Creating build environment for platform esp32/wt32_eth01.
   Working directory: /home/pi/.node-red/mcu-plugin-cache/t5rg16s07b
   ...
   >> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
   >> mcconfig -d -x localhost:5004 -l -m -p esp32/wt32_eth01 -t build ssid=xxx password=yyy
   /home/pi/esp-idf/tools/idf.py
   ...
   Creating esp32 image...
   Merged 25 ELF sections
   Successfully created esp32 image.
   ...
    /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/xsProj-esp32/build/xs_esp32.bin
   xs_esp32.bin binary size 0x1710a0 bytes. Smallest app partition is 0x390000 bytes. 0x21ef60 bytes (60%) free.
   Executing action: all (aliases: build)
   Running ninja in directory /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/xsProj-esp32/build
   Executing "ninja all"...
   Project build complete. 
   ```
3. Then I choose build target ***"deploy"***, which again seems to be successfull:
   ```
   Starting build process...
   ...
   >> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
   >> mcconfig -d -x localhost:5004 -l -m -p esp32/wt32_eth01 -t deploy ssid=xxx password=yyy
   /home/pi/esp-idf/tools/idf.py
    # uploading to esp32
   ...
   esptool.py v3.3.2
   Serial port /dev/ttyUSB0
   Connecting....
   Chip is ESP32-D0WD (revision 1)
   ...
   Wrote 1511584 bytes (899636 compressed) at 0x00010000 in 15.1 seconds (effective 800.7 kbit/s)...
   Hash of data verified.
   Compressed 3072 bytes to 123...
   Writing at 0x00008000... (100 %)
   Wrote 3072 bytes (123 compressed) at 0x00008000 in 0.1 seconds (effective 326.3 kbit/s)...
   Hash of data verified.
   ...
   Done
   ```
4. Next I leave the programming mode of the WT32-ETH01, by disconnecting the IO0 pin from the GND.
5. Finally I choose build target ***"xsbug"***, but that build fails:
   ```
   >> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
   >> mcconfig -d -x localhost:5004 -l -m -p esp32/wt32_eth01 -t xsbug ssid=xxx password=yyy
   /home/pi/esp-idf/tools/idf.py
   Terminated
   ```
   Note that my debug target is ***"xsbug-log"*** because I have a headless Raspberry Pi.

**Expected behavior**
I think I am almost at the end of the setup, so would appreciate if somebody could have a look why this fails.  I am not sure if there are any other logs that might help to troubleshoot this?  If you need any extra infromation, don't hesitate to ask!

Thanks!!
Bart

andycarle, 2023-02-18:
If you run `mcconfig` with the `-v` flag, does it give you any more information? 

I'm just trying to sort out what path `mcconfig` is taking through the Makefile. From your console output, it looks like it's running `idf.py`, which shouldn't be the case for `-t xsbug`.

bartbutenaers, 2023-02-18:
Since I want to stick to Ralph's plugin, I have update his sidebar code to add an extra `-v` parameter to the command. 
But that doesn't give extra output.  This is the full log:
```
Starting build process...
Host system check: #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023
MCU Build system check: p1.1.2-post + #8224a11 @ m3.6.0-104-g819c387
HOME directory check: /home/pi
Creating build environment for platform esp32/wt32_eth01.
Working directory: /home/pi/.node-red/mcu-plugin-cache/t5rg16s07b
$IDF_PATH is defined: /home/pi/esp-idf
$IDF_PYTHON_ENV_PATH is not defined.
> cd /home/pi/.node-red/mcu-plugin-cache/t5rg16s07b
UPLOAD_PORT = /dev/ttyUSB0
Creating build script file...
> /bin/bash ./build.sh
>> source /home/pi/esp-idf/export.sh
Detecting the Python interpreter
Checking "python" ...
Python 3.9.2
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /home/pi/.espressif/python_env/idf4.4_py3.9_env/bin/python
Checking if Python packages are up to date...
Python requirements from /home/pi/esp-idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /home/pi/esp-idf/components/esptool_py/esptool
  /home/pi/esp-idf/components/espcoredump
  /home/pi/esp-idf/components/partition_table
  /home/pi/esp-idf/components/app_update
  /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /home/pi/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /home/pi/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /home/pi/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /home/pi/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/pi/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
  /home/pi/.espressif/python_env/idf4.4_py3.9_env/bin
  /home/pi/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

>> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
>> mcconfig -d -x localhost:5004 -l -v -m -p esp32/wt32_eth01 -t xsbug ssid=xxx password=yyy
/home/pi/esp-idf/tools/idf.py
Terminated
```

andycarle, 2023-02-21:
@bartbutenaers Understood. I'm working through a setup as close as I can get to yours today to try to replicate this issue. I'll report back on what I learn.

andycarle, 2023-02-22:
@bartbutenaers, I think I understand what's going on here. Could you please test something for me and see if it helps? If it does, I'll work on a more proper fix.

In your local clone of the Moddable SDK, could you please modify [`$MODDABLE/tools/mcconfig/make.esp32.mk`](https://github.com/Moddable-OpenSource/moddable/blob/3c3f72f2b12775481e2962222eeb2dfc2321890a/tools/mcconfig/make.esp32.mk#L420). At line 420 (just before `all: precursor`) please add one line:

```makefile
KILL_XSBUG = 
```

This will remove a command intended to kill any running instances of `xsbug`. But since you're not running `xsbug` locally, that shouldn't be a problem.

Please give that a shot and see if it helps you continue making progress.

bartbutenaers, 2023-02-23:
@andycarle,

You nailed it.  I am very impressed that you solved such kind of issues this fast...

Now I get this when I hit the "Build" button
```
....
idf.py build

>> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
>> mcconfig -d -x localhost:5004 -l -v -m -p esp32/wt32_eth01 -t xsbug ssid=checkpointCharly password=cbutpHvrr8
/home/pi/esp-idf/tools/idf.py
# starting xsbug
PORT_USED=$(grep 'Serial port' /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/xsProj-esp32/flashOutput | awk 'END{print($3)}'); \
cd /home/pi/Projects/moddable/tools/xsbug-log && node xsbug-log bash -c \"XSBUG_PORT=5004 XSBUG_HOST=localhost serial2xsbug /dev/ttyUSB0 460800 8N1\"
# xsbug-log listening on port 5002. ^C to exit.
```
And after restarting my wt32-eth01, my Node-RED flow is running on it.
This is really awesome.  

You can close this issue if you don't need further input from me.
And please don't forget to add that magic statement to the official makefile ;-)

Thank you so much for your patience and support.
I am pretty sure we will meet again soon, when I have time to test your new sensor types...

Kind regards,
Bart

bartbutenaers, 2023-02-23:
@andycarle,

You nailed it. I am very impressed that you solved such kind of issues this fast...

Now I get this when I hit the "Build" button

....
idf.py build

>> IDF_PYTHON_ENV_PATH: /home/pi/.espressif/python_env/idf4.4_py3.9_env
>> mcconfig -d -x localhost:5004 -l -v -m -p esp32/wt32_eth01 -t xsbug ssid=xxx password=yyy
/home/pi/esp-idf/tools/idf.py
# starting xsbug
PORT_USED=$(grep 'Serial port' /home/pi/Projects/moddable/build/tmp/esp32/wt32_eth01/debug/t5rg16s07b/xsProj-esp32/flashOutput | awk 'END{print($3)}'); \
cd /home/pi/Projects/moddable/tools/xsbug-log && node xsbug-log bash -c \"XSBUG_PORT=5004 XSBUG_HOST=localhost serial2xsbug /dev/ttyUSB0 460800 8N1\"
# xsbug-log listening on port 5002. ^C to exit.
And after restarting my wt32-eth01, my Node-RED flow is running on it.
This is really awesome.

You can close this issue if you don't need further input from me.
And please don't forget to add that magic statement to the official makefile ;-)

Thank you so much for your patience and support.
I am pretty sure we will meet again soon, when I have time to test your new sensor types...

Kind regards,

bartbutenaers, 2023-04-05:
I am going to close this, because it all works fine now.  Thanks a lot !!!

closed 2023-04-05 by bartbutenaers

phoddie, 2023-04-05:
Glad we sorted this out. Thanks for raising the issue.