Moddable-OpenSource/moddable #1043: Is it a loop related performance issue?

zjyyyyy123, 2023-02-21:
**Build environment:** Linux 18.04.1-Ubuntu
**Target device:**  moddable-3.7.0-release[https://github.com/Moddable-OpenSource/moddable/archive/refs/tags/3.7.0.zip]

```shell
MODDABLE="$(pwd)/moddable"
export MODDABLE
cd $MODDABLE/xs/makefiles/lin
make
```

**Description**
When I interpret the **A.js** in six different engines, the running time of XS is more than ten times higher than that of other engines. After looking at the test case, I found that it contains multiple loops.

Does this kind of situation belong to the **performance issue** of XS for loop processing?

**Steps to Reproduce**

use command :

```shell
    echo "=================hermes_release=================="
    time ./engines/hermes/hermes-0.12.0/build_release/bin/hermes -w A.js
    echo "================== by quickjs-2021-03-27========================"
    time ./engines/quickjs/quickjs-2021-03-27/qjs A.js
    echo "================== by jerryscript-2.4.0========================"
    time ./engines/jerryscript/jerryscript-2.4.0/jerryscript-2.4.0/build/bin/jerry A.js
    echo "================== by moddable-3.7.0========================"
    time ./engines/XS/moddable-3.7.0/build/bin/lin/release/xst A.js
    echo "================== by duktape-2.7.0========================"
    time ./engines/duktape/duktape-2.7.0/duk A.js
    echo "================== by mujs-1.3.2========================"
    time ./engines/mujs/mujs-1.3.2/build/release/mujs A.js
```

**Test case**

```javascript
A.JS
var NISLFuzzingFunc = function (dictionary) {
    let listHtml = '<ul>';
    for (var INDEX = 0; INDEX < 1000; INDEX++) {
        for (let key in dictionary) {
            let value = dictionary[key];
            listHtml += `<li>`;
            if (value)
                listHtml += '<li style="list-style-type:none"><ul>';
            if (value)
                listHtml += '</li></ul>';
        }
    }
    return listHtml;
};
var NISLParameter0 = [
    true,
    true,
    true,
    true,
    true
];
var NISLCallingResult = NISLFuzzingFunc(NISLParameter0);
NISLCallingResult

```

**Expected behavior**
	My expected behavior that the six engines run at about the same time,But in reality, the XS engine runs the longest of the six engines.

**Actual behavior**

```shell
=================hermes_release==================

real    0m0.009s
user    0m0.009s
sys     0m0.000s
================== by quickjs-2021-03-27========================

real    0m0.325s
user    0m0.124s
sys     0m0.201s
================== by jerryscript-2.4.0========================

real    0m0.030s
user    0m0.030s
sys     0m0.000s
================== by moddable-3.7.0========================

real    0m0.579s
user    0m0.567s
sys     0m0.012s
================== by duktape-2.7.0========================
SyntaxError: unterminated statement (line 2)
    at [anon] (/home/yhy/workspace/specious_testcases/10.js:2) internal
    at [anon] (duk_js_compiler.c:6826) internal
error in executing file /home/yhy/workspace/specious_testcases/10.js

real    0m0.001s
user    0m0.001s
sys     0m0.000s
================== by mujs-1.3.2========================
SyntaxError: /home/yhy/workspace/specious_testcases/10.js:2: unexpected token: (identifier) (expected ';')

real    0m0.001s
user    0m0.001s
sys     0m0.000s
```

phoddie, 2023-02-21:
Thank you for sharing your findings. Unfortunately, synthetic performance tests are not representative of real-world use of XS.

> My expected behavior that the six engines run at about the same time

This is a naive premise.

closed 2023-02-21 by phoddie

zjyyyyy123, 2023-02-21:
What I want to know is that different engines run the same test case, but the execution time of xs is slower than other engines, is it possible to think that xs engine performance is slightly weaker than other engines when processing loops here?
Also, based on your reply, I would like to ask, what kind of situation will be considered as a performance difference?thank u ：)

phoddie, 2023-02-21:
Apologies, but you don't seem to have a good understanding of the topic.