Moddable-OpenSource/moddable #1045: Hide wifi credentials in log by default

bartbutenaers, 2023-02-23:
Dear,

It would be nice if the wifi username and password could be hided by default in the log.

Currently you get this:
```
>> mcconfig -d -x localhost:5004 -l -v -m -p esp32/wt32_eth01 -t xsbug ssid=my_wifi_ssid password=my_wifi_password
```

Which means you share your wifi credentials with the entire world, as soon as your share your logs to get support somewhere (here on Github, Node-RED Discourse forum, ...).  It is very easily forgotten to remove it from the logs before sharing it. 

I can imagine that it is useful to show the wifi credentials in the logs (e.g. via a command line parameter) in some use cases, but not by default imho.

Thank you for taking this into consideration!!

Bart

phoddie, 2023-02-23:
The Moddable SDK doesn't put your credentials in any log. What you are showing is a command line invoking the `mcconfig` command line tool in the Moddable SDK. There are alternatives to passing your credentials directly on the command line:

- Put your credentials into environment variables, such as $WIFIPASSWORD. Then with mcconfig instead of `ssid=my_wifi_ssid` use `ssid=$WIFIPASSWORD`.
- Put the credentials in the `config` section of the [project manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#config)

closed 2023-02-23 by phoddie

tve, 2023-02-23:
Bart, was this in the context of node-red-mcu? If so, you perhaps opened an issue in the wrong repo?

bartbutenaers, 2023-02-25:
@tve ,
Yes indeed.  I enter my credentials in the Node-RED sidebar, but I wasn't aware that the sidebar was responsible for generating the commands.  I will create a new issue in Ralph's repo.
Thanks!