Moddable-OpenSource/moddable #1051: ESP SPI incorrectly defaults select pin to active=1

tve, 2023-02-27:
**Build environment:** Linux
**Target device:** ESP32

**Description**
The SPI implementation for ESP incorrectly defaults the select pin active state to 1, see https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/spi/esp/spi.c#L66
According to ECMA-419: "active: The value to write to the select pin when the SPI instance is active. Must be 1 or 0. This property is optional and defaults to 0."

phoddie, 2023-02-27:
Thanks for the report.

> The SPI implementation for ESP incorrectly defaults the select pin active state to 1

True. The fix is trivial but updating callers that depend on the current implementation is some work.

phoddie, 2023-04-26:
Fixed with the March 1 commit noted above.

closed 2023-04-26 by phoddie