Moddable-OpenSource/moddable #1062: Unable to set non-default port on xsbug-log

bartbutenaers, 2023-03-13:
**Build environment:** Linux on Raspberry
**Target device:** wt32-eth01

**Description**

@ralphwetzel was [investigating ](https://discourse.nodered.org/t/node-red-mcu-plugin-v1-0-integrating-node-red-mcu-edition-into-the-node-red-editor/72693/172?u=bartbutenaers)an issue for me, and he found out that it is not possible to set a custom port (5004) on xsbug-log via the "-x" parameter.  As you can see in his log, xsbug-log keeps listening to the default port (5002):
```
>> IDF_PYTHON_ENV_PATH: /home/ralph/.espressif/python_env/idf4.4_py3.10_env
>> mcconfig -d -x localhost:5004 -l -m -p esp32/wt32_eth01 -t xsbug
/home/ralph/esp32/esp-idf/tools/idf.py
# starting xsbug
# xsbug-log listening on port 5002. ^C to exit
```

**Steps to Reproduce**

1. Run the above command
2. See in the log that it starts listening to the default port

**Expected behavior**

That xsbug-log starts listening at the port specified with the "-x" parameter.

Thanks!!
Bart

phoddie, 2023-03-13:
Thank you for the report.

I successfully tested a fix on macOS. Perhaps you can try the parallel change on your Linux set-up? Change [line 399 of make.esp32.mk](https://github.com/Moddable-OpenSource/moddable/blob/746d1378311d1c55157578b8d5304bc46972457b/tools/mcconfig/make.esp32.mk#LL399) to:

```
DO_LAUNCH := cd $(MODDABLE)/tools/xsbug-log && XSBUG_PORT=$(XSBUG_PORT) node xsbug-log $(LOG_LAUNCH)
```

bartbutenaers, 2023-03-13:
Evening @phoddie,

Thanks a lot for your fast support!!!
I can indeed confirm that (after applying your fix) xsbug-log now indeed start listening on port 5004 via Ralph's sidebar.
So you can close this issue!

Bart

phoddie, 2023-03-13:
Very nice! Thank you for the quick confirmation. (the changes will be included in our next Moddable SDK update)

closed 2023-03-13 by phoddie