Moddable-OpenSource/moddable #1064: "FT6202: bad FT6206_REG_VENDID!" on the current M5Stack Core2 revision

meganetaaan, 2023-03-16:
**Build environment:** Any OS
**Target device:** esp32/m5stack_core2

**Description**
I am reporting this issue because changes on the Moddable side are required for the issue in https://github.com/meganetaaan/stack-chan/issues/135. Due to a specification change in the touch control chip for M5Stack Core2, the value of VENDID has been changed from 17 to 1.

**Steps to Reproduce**

1. Build and install any app using: mcconfig -m -d -p esp32/m5stack_core2
2. See error
```
/Users/apple/dev/moddable/modules/drivers/ft6206/modFt6206.c (131) # Break: FT6202: bad FT6206_REG_VENDID!
```

**Expected behavior**
The app should launch without errors.

**Other information**

I have received a report that the issue was resolved after modifying the relevant code to
```
if ((1 != data[0]) && (17 != data[0])) xsUnknownError("bad FT6206_REG_VENDID");
```
However, it is necessary to consider whether this workaround is appropriate as a driver for FT6202.

phoddie, 2023-03-16:
Interesting. Thank you for the report. Do you know what the part number is for the chip? I'd like to see a data sheet.

meganetaaan, 2023-03-16:
>However, it is necessary to consider whether this workaround is appropriate as a driver for FT6202.
>FT6202

Oops. This is my typo of FT6206. Please forget it.

The reported issue is due to a silent breaking change, as there have been no changes to [the specs of the M5Stack Core2, which currently uses FT6336u](http://docs.m5stack.com/en/core/core2). According to one of the developers of M5Stack, the new chip is compatible at the application level except for the vendor ID. The default value of the vendor ID in the datasheet is 0x11, which differs from both the old value of 0x17 and the new value of 0x01. While it may seem ad hoc, I believe that treating 0x01 as a valid ID, similar to the aforementioned workaround, would be sufficient in this case.

meganetaaan, 2023-03-16:
Ah, by the way, @Fuji-P, who reported this workaround, is positive about contributing code. It would be greatly appreciated if you could check the pull request later.

phoddie, 2023-03-16:
Thank you for the clarifications. I wonder why the vendor ID changed. Perhaps this part is a clone. In any case, I agree that we should support this configuration.

FWIW– the vendor ID that our implementations check for appears to be consistently `17` which is equal to the datasheet value of `0x11`.

I see the PR from @Fuji-P. Always happy to welcome a new contributor. We'll get that taken care of.

phoddie, 2023-03-17:
This has been [merged](https://github.com/Moddable-OpenSource/moddable/commit/f113f476982cebb2ba799dc017269eb3e812d396). The corresponding changes have been [applied](https://github.com/Moddable-OpenSource/moddable/commit/91075268b1e1836cab1653d331ee6e5eb3a2a927) to the ECMA-419 version of the FT6202 driver.

closed 2023-03-17 by phoddie