Moddable-OpenSource/moddable #1071: Error in makefiles created by Node-RED MCU

NextWorldLab, 2023-03-25:
**Build environment:** Windows　Node-RED MCU Plugin 
**Target device:** 　M5Stich_cplus + sht3x

**Description**
Thank you for automatically embedding the sensor driver.
I get the following error:
**************************
mcconfig -d -x localhost:5004 -m -p esp32/m5stick_cplus
C:\Espressif\tools\idf-exe\1.0.3\idf.py.exe
C:\Espressif\frameworks\esp-idf-v4.4.4\tools\idf.py
c:\pjt\moddable\build\tmp\esp32\m5stick_cplus\debug\qnkkmq7msvl\makefile(111) : fatal error U1033: 構文エラー : 予期しない 'c:\pjt\moddable\modules\drivers\sensors\sht3x\manifest.json'
Stop.
**************************
Corresponding part
[image]

I think "#3" in the previous line is probably bad.

thanks.

phoddie, 2023-03-25:
Interesting. I agree that the #3 is a problem. Would you please provide the exported flow?Thank you. -- Peter

NextWorldLab, 2023-03-25:
flows.json is this
```
[
  {
    "id": "959d95b9238f454e",
    "type": "tab",
    "label": "フロー 3",
    "disabled": false,
    "info": "",
    "env": []
  },
  {
    "id": "ca0d5fae48fbbf69",
    "type": "inject",
    "z": "959d95b9238f454e",
    "name": "",
    "props": [
      {
        "p": "payload"
      },
      {
        "p": "topic",
        "vt": "str"
      }
    ],
    "repeat": "5",
    "crontab": "",
    "once": false,
    "onceDelay": 0.1,
    "topic": "",
    "payload": "",
    "payloadType": "date",
    "x": 190,
    "y": 120,
    "wires": [
      [
        "39a8b30d5a307223"
      ]
    ]
  },
  {
    "id": "cc3f24157c2ed3d8",
    "type": "debug",
    "z": "959d95b9238f454e",
    "name": "debug 1",
    "active": true,
    "tosidebar": true,
    "console": false,
    "tostatus": false,
    "complete": "false",
    "statusVal": "",
    "statusType": "auto",
    "x": 500,
    "y": 120,
    "wires": []
  },
  {
    "id": "39a8b30d5a307223",
    "type": "sensor",
    "z": "959d95b9238f454e",
    "name": "",
    "platform": "",
    "module": "embedded:sensor/Humidity-Temperature/SHT3x",
    "options": {
      "sensor": {
        "io": "I2C",
        "bus": "default",
        "address": "0x44"
      }
    },
    "configuration": "{}",
    "moddable_manifest": {
      "include": [
        "$(MODDABLE)/modules/drivers/sensors/sht3x/manifest.json"
      ]
    },
    "x": 370,
    "y": 120,
    "wires": [
      [
        "cc3f24157c2ed3d8"
      ]
    ]
  }
]
```

NextWorldLab, 2023-03-25:
This is build.bat generated by mcu plugin.
```
CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"
pushd %IDF_PATH%
CALL "C:\Espressif\idf_cmd_init.bat"
popd
@echo mcconfig -d -x localhost:5004 -m -p esp32/m5stick_cplus
mcconfig -d -x localhost:5004 -m -p esp32/m5stick_cplus
```

phoddie, 2023-03-25:
Thank you for the additional information. The extra data on the source path is unnecessary. It can be removed by replacing [this line in mcmanifest.js](https://github.com/Moddable-OpenSource/moddable/blob/06db76e142e532266f4da5748e81e0e93139a7dc/tools/mcmanifest.js#L1756) with the following:

```js
this.parseManifest(source, {...node.moddable_manifest, directory: this.currentDirectory});
```
Would you give that a try? 

FWIW – The path with extra data is being output for all platforms. However, it doesn't generate an error on macOS but clearly does on Windows. That's why I didn't notice the problem.

NextWorldLab, 2023-03-26:
Thank you for your reply.

Once fixed, I get this error:
"### Error: 'C:\Users\tauchi\node_modules\\\@ralphwetzel\node-red-mcu-plugin\node-red-mcu\manifest_runtime.json': manifest not found!"

Probably, I think that it is really "c:\users\tauchi\\.node-red\node_modules\.....", but there is no ".node-red".

The manifest automatically generated by the plugin is as follows.
```
{
  "build": {
    "MODULES": "c:\\pjt\\moddable\\modules",
    "MCUROOT": "C:\\Users\\tauchi\\.node-red\\node_modules\\@ralphwetzel\\node-red-mcu-plugin\\node-red-mcu",
    "REDNODES": "C:\\Users\\tauchi\\AppData\\Roaming\\npm\\node_modules\\node-red\\node_modules\\@node-red\\nodes"
  },
  "include": [
    "$(MCUROOT)/manifest_host.json",
    "$(MCUROOT)/./nodes/mcu/sensor/manifest.json"
  ],
  "modules": {
    "*": [
      "./main",
      {
        "source": "./flows",
        "transform": "nodered2mcu"
      }
    ],
    "~": []
  },
  "preload": [
    "flows"
  ],
  "config": {
    "noderedmcu": {
      "editor": true
    }
  }
}
```

NextWorldLab, 2023-03-26:
sorry.

Rebuilding Moddeable and restarting the PC seems to be fine.
It seems that there are other problems, but I will give them in another issue.

thank you.

closed 2023-03-26 by NextWorldLab

phoddie, 2023-03-26:
Thanks for confirming. I should have mentioned that is necessary to rebuild the tools after making that change. Apologies.