Moddable-OpenSource/moddable #1072: -x command broken

ralphwetzel, 2023-03-26:
The -x command @ the `mcconfig` command line currently has no effect.
The MCU talks (directly) to xsbug.

Example command:
`mcconfig -d -x localhost:5004 -m -p sim/moddable_two -t all`

This happens on macOS, [Linux](https://github.com/phoddie/node-red-mcu/discussions/105#discussioncomment-5429161) & as well [Windows](https://github.com/ralphwetzel/node-red-mcu-plugin/issues/34).  

The last config I can confirm fully operational was:

```
    26 Mar 09:31:14 - [info] Node-RED MCU Edition Runtime Version: #942132f
    26 Mar 09:31:14 - [info] Node-RED MCU Edition Plugin  Version: v1.3.0
    26 Mar 09:31:14 - [info] Moddable SDK Version: v3.7.6-46-g2a4641e
```

which is the
* current version of the `node-red-mcu-plugin`
* the current version of `node-red-mcu`
* [Moddable SDK from March 21](https://github.com/Moddable-OpenSource/moddable/commit/2a4641e93239b371265bda4b588a7c2e718c67cf)

phoddie, 2023-03-26:
Thanks for the report. The line to export XSBUG_PORT was accidentally lost in recent changes. It will be back shortly. 

FYI – it belongs here in mcmanifest.js, just after setting XSBUG_HOST.

```js
		this.line("XSBUG_HOST = ", tool.xsbug?.host ?? "localhost");
		this.line("XSBUG_PORT = ", tool.xsbug?.port ?? 5002);
```

bartbutenaers, 2023-03-26:
@phoddie,
Thanks for sharing the fix so rapidly.  I have added that line manually, and I can confirm I now finally see the same [debug info](https://github.com/phoddie/node-red-mcu/discussions/105#discussioncomment-5425475) as @ralphwetzel.  Again a step closer :-)

ralphwetzel, 2023-03-26:
👍 Thank you for the timely fix.

closed 2023-03-26 by ralphwetzel