Moddable-OpenSource/moddable #1078: Windows makefile Syntax Error

NextWorldLab, 2023-04-05:
**Build environment:**  Windows (Node-Red-MCU Plugin)
**Target device:** m5stick_cplus

**Description**
Build stops with a syntax error.

**Steps to Reproduce**
build.bat generated by plugin.
```
CALL "C:\Program Files\Microsoft Visual Studio\2022\Community\VC\Auxiliary\Build\vcvars32.bat"
pushd %IDF_PATH%
CALL "C:\Espressif\idf_cmd_init.bat"
popd
@echo mcconfig -d -x localhost:5004 -m -p esp32/m5stick_cplus -r 180
mcconfig -d -x localhost:5004 -m -p esp32/m5stick_cplus -r 180
```

3. See error
[image]

**Other information**
Looking at the makefile, it looks like C_FLAGS_SUBPLATFORM is not defined.

phoddie, 2023-04-05:
Should be better now. Would you please confirm?

NextWorldLab, 2023-04-05:
Now it works.
thank you.

closed 2023-04-05 by NextWorldLab

phoddie, 2023-04-05:
Excellent. Sorry for the trouble.