Moddable-OpenSource/moddable #1084: No possiblity to exchange data with AWS IoT

PatrykStachura, 2023-04-11:
**Build environment:** Linux Ubuntu
**Target device:** NodeMCU ESP32

**Description**
I'm trying to achive exchange data between my IoT device and AWS IoT Core service. There's everything configured as it is described in docs and compared to your examples. I've get Device Cert, CA RSA 2048 Cert & PrivateKey Cert compiled into `.der` files as it is in `Readme.md` file inside aws directory. The thing is i achive an connection, but there's no option to exchange data between service and device. The problem like this (connectivity achive, but no possibility to exchange data) appears when there's something with `Cert`. I stuck a little bit here, because i'm not aware how it is compiled to `.der` files and if that part is working as it suposed to. Would be lovely if i've get some answear from you.  Also, there's no existing function for switch `CONNACK` 

xsbug compiler message:

[image]

main code for aws module:
[image]

forgive me grammar,
Best regards

phoddie, 2023-04-12:
Thanks for posting the issue. @dabdoue and I have been discussing the report. We'll get a response to you shortly.

dabdoue, 2023-04-12:
@PatrykStachura, @phoddie and I believe that your error is with the name of the function you have called `onConnect()` in `SecureClient`. Can you try renaming that function to `onReady()`? The simulator is a quick and easy way to test this as well.

PatrykStachura, 2023-04-14:
@dabdoue Yes, my friend it was the usage of wrong methond on my side. Can be closed. Appraciate your time!

closed 2023-04-14 by PatrykStachura