Moddable-OpenSource/moddable #1086: [ESP-IDF 5] build error in function 'modInstallMods'

stc1988, 2023-04-12:
**Moddable:**  b042dc8723af4dbdd66d840be632812256bf6e99
**Build environment:** macOS,
**Target device:** eps32/m5stack

**Steps to Reproduce**
Build `examples/piu/outline/figures` with `mcconfig -d -m -p esp32/m5stack`

**Other information**

```
mcconfig -d -m -p esp32/m5stack
/Users/satoshi/.local/share/esp32/esp-idf/tools/idf.py
/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/makefile:1357: warning: overriding commands for target `/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/ftimage.h.xsi'
/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/makefile:1288: warning: ignoring old commands for target `/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/ftimage.h.xsi'
# xsc pins/digital.xsb
# xsc ili9341.xsb
# xsc m5button.xsb
# xsc time.xsb
# xsc Resource.xsb
# xsc timer.xsb
# xsc main.xsb
# xsc modules.xsb
# xsc commodetto/Bitmap.xsb
# xsc pins/digital/monitor.xsb
# xsc pins/audioout.xsb
# xsc setup/target.xsb
# xsc commodetto/Poco.xsb
# xsc m5stack/screen.xsb
# xsc commodetto/parseRLE.xsb
# xsc commodetto/parseBMP.xsb
# xsc piu/All.xsb
# xsc piu/Timeline.xsb
# xsc piu/CombTransition.xsb
# xsc piu/WipeTransition.xsb
# xsc piu/MC.xsb
# xsc setup/piu.xsb
# xsc commodetto/outline.xsb
# xsc commodetto/pocoOutline.xsb
# xsc piu/shape.xsb
# xsc mc/config.xsb
# xsid digital.c.xsi
# xsid modGPIO.c.xsi
# xsid modSPI.c.xsi
# xsid modIli9341.c.xsi
# xsid modMonitor.c.xsi
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
# xsid Resource.c.xsi
# xsid modInstrumentation.c.xsi
# xsid timer.c.xsi
# xsid modPreference.c.xsi
# xsid modLwipSafe.c.xsi
# xsid piuFont.c.xsi
# xsid piuDie.c.xsi
# xsid piuImage.c.xsi
# xsid piuRegion.c.xsi
# xsid piuView.c.xsi
# xsid ftgrays.c.xsi
# xsid ftcalc.c.xsi
# xsid fttrigon.c.xsi
# xsid ftstroke.c.xsi
# xsid piuTexture.c.xsi
# xsid ftraster.c.xsi
# xsid modules.c.xsi
# xsid commodettoBitmap.c.xsi
# xsid ftutil.c.xsi
# xsid dvi_adpcm_decode.c.xsi
# xsid audioout.c.xsi
# xsid sbc_decoder.c.xsi
# xsid commodettoPocoBlit.c.xsi
[... 171 lines cut ...]
/Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.c:1335:9: error: unknown type name 'spi_flash_mmap_handle_t'
 1335 |         spi_flash_mmap_handle_t handle;
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.c:1340:83: error: 'SPI_FLASH_MMAP_DATA' undeclared (first use in this function)
 1340 |                 if (ESP_OK == esp_partition_mmap(gPartition, 0, gPartition->size, SPI_FLASH_MMAP_DATA, (const void **)&gPartitionAddress, &handle)) {
      |                                                                                   ^~~~~~~~~~~~~~~~~~~
/Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.c:1340:83: note: each undeclared identifier is reported only once for each function it appears in
/Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.c:1340:139: warning: passing argument 6 of 'esp_partition_mmap' from incompatible pointer type [-Wincompatible-pointer-types]
 1340 |                 if (ESP_OK == esp_partition_mmap(gPartition, 0, gPartition->size, SPI_FLASH_MMAP_DATA, (const void **)&gPartitionAddress, &handle)) {
      |                                                                                                                                           ^~~~~~~
      |                                                                                                                                           |
      |                                                                                                                                           int *
In file included from /Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.h:515,
                 from /Users/Satoshi/Projects/moddable/xs/platforms/esp/xsPlatform.h:116,
                 from /Users/Satoshi/Projects/moddable/xs/sources/xsCommon.h:41,
                 from /Users/Satoshi/Projects/moddable/xs/sources/xsAll.h:41,
                 from /Users/Satoshi/Projects/moddable/xs/platforms/esp/xsHost.c:38:
/Users/satoshi/.local/share/esp32/esp-idf/components/esp_partition/include/esp_partition.h:375:81: note: expected 'esp_partition_mmap_handle_t *' {aka 'long unsigned int *'} but argument is of type 'int *'
  375 |                              const void** out_ptr, esp_partition_mmap_handle_t* out_handle);
      |                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/xsPlatform.c.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack/debug/figures/xsHost.c.o] Error 1
### 647 instances, 1108 keys, 93 colors, 0 holes
```

phoddie, 2023-04-12:
Thanks for trying out the IDF 5 support! There was some change in where `SPI_FLASH_SEC_SIZE` is defined in IDF 5. I thought we caught all of those, but obviously not. I'll clean that up.

phoddie, 2023-04-13:
Excellent, thank you!