Moddable-OpenSource/moddable #1088: [ESP-IDF 5]build error  in function 'xs_servo

stc1988, 2023-04-13:
**Moddable:**  **Moddable:**  b042dc8723af4dbdd66d840be632812256bf6e99
**Build environment:** macOS,
**Target device:** eps32/m5stack_core2

**Steps to Reproduce**
Build `examples/pins/servo` with `mcconfig -d -m -p esp32/m5stack_core2`

**Other information**

```
mcconfig -d -m -p esp32/m5stack_core2
/Users/satoshi/.local/share/esp32/esp-idf/tools/idf.py
# xsc pins/digital.xsb
# xsc ili9341.xsb
# xsc preference.xsb
# xsc ft6206.xsb
# xsc mpu6886.xsb
# xsc M5StackCoreTouch.xsb
# xsc axp192.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc main.xsb
# xsc commodetto/Bitmap.xsb
# xsc pins/i2c.xsb
# xsc setup/target.xsb
# xsc pins/audioout.xsb
# xsc pins/smbus.xsb
# xsc pins/servo.xsb
# xsc mc/config.xsb
# xsid digital.c.xsi
# xsid modIli9341.c.xsi
# xsid modGPIO.c.xsi
# xsid modSPI.c.xsi
# xsid modI2C.c.xsi
# xsid modFt6206.c.xsi
# xsid modPreference.c.xsi
# xsid modTime.c.xsi
# xsid timer.c.xsi
# xsid modTimer.c.xsi
# xsid modInstrumentation.c.xsi
# xsid Resource.c.xsi
# xsid modLwipSafe.c.xsi
# xsid modServo.c.xsi
# xsid commodettoBitmap.c.xsi
# xsid i2c.c.xsi
# xsid modSPI.h.xsi
# xsid modGPIO.h.xsi
# xsid dvi_adpcm_decode.c.xsi
# xsid modI2C.h.xsi
# xsid sbc_decoder.c.xsi
# xsid modPreference.h.xsi
# xsid audioout.c.xsi
# xsid modInstrumentation.h.xsi
# xsid modTimer.h.xsi
# xsid commodettoBitmap.h.xsi
# xsid modLwipSafe.h.xsi
# xsid sbc_decoder.h.xsi
# wav2maud bflatmajor.maud
# Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32/build/config/sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=1500000 -DIDF_TARGET=esp32 -DESP32_SUBCLASS=esp32 -DCCACHE_ENABLE=0 /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32"...
Loading defaults file /Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32/sdkconfig.mc...
-- Configuring done (1.8s)
-- Generating done (0.2s)
-- Build files have been written to: /Users/satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/xsProj-esp32/build
# cc xsAll.c (strings in flash)
# cc xsAPI.c (strings in flash)
# cc xsArguments.c (strings in flash)
# cc xsAtomics.c (strings in flash)
# cc xsBoolean.c (strings in flash)
# cc xsArray.c (strings in flash)
# cc xsBigInt.c (strings in flash)
# cc xsDataView.c (strings in flash)
# cc xsDebug.c (strings in flash)
# cc xsCommon.c (strings in flash)
# cc xsCode.c (strings in flash)
# cc xsDate.c (strings in flash)
# cc xsError.c (strings in flash)
# cc xsGenerator.c (strings in flash)
# cc xsGlobal.c (strings in flash)
# cc xsFunction.c (strings in flash)
# cc xsJSON.c (strings in flash)
# cc xsMapSet.c (strings in flash)
# cc xsLexical.c (strings in flash)
# cc xsMarshall.c (strings in flash)
# cc xsMemory.c (strings in flash)
# cc xsMath.c (strings in flash)
# cc xsModule.c (strings in flash)
# cc xsObject.c (strings in flash)
# cc xsPromise.c (strings in flash)
# cc xsNumber.c (strings in flash)
# cc xsProperty.c (strings in flash)
# cc xsProxy.c (strings in flash)
# cc xsRun.c (strings in flash)
# cc xsRegExp.c (strings in flash)
# cc xsScope.c (strings in flash)
# cc xsScript.c (strings in flash)
# cc xsSymbol.c (strings in flash)
# cc xsSourceMap.c (strings in flash)
# cc xsString.c (strings in flash)
# cc xsSyntaxical.c (strings in flash)
# cc xsTree.c (strings in flash)
# cc xsdtoa.c (strings in flash)
# cc xsType.c (strings in flash)
# cc xsre.c (strings in flash)
# cc xsmc.c (strings in flash)
# cc e_pow.c (strings in flash)
# cc xsPlatform.c (strings in flash)
# cc xsHost.c (strings in flash)
# cc xsHosts.c (strings in flash)
# xsl modules
# mcrez resources
### 392 instances, 772 keys, 63 colors, 0 holes
Total resource size: 21078 bytes
### warning: axp192: default.CHARGE_CURRENT: not frozen
### warning: setup/target: default() state: not frozen
### warning: globalThis.Host: not frozen
# cc mc.xs.c (slots in flash)
# cc digital.c.o
# cc modGPIO.c.o
# cc modSPI.c.o
# cc modIli9341.c.o
# cc modI2C.c.o
# cc modPreference.c.o
# cc modFt6206.c.o
# cc modTime.c.o
# cc modTimer.c.o
# cc timer.c.o
# cc Resource.c.o
# cc modInstrumentation.c.o
# cc modLwipSafe.c.o
# cc modServo.c.o
# cc commodettoBitmap.c.o
# cc i2c.c.o
# cc audioout.c.o
# cc sbc_decoder.c.o
# cc dvi_adpcm_decode.c.o
/Users/satoshi/Projects/moddable/modules/pins/servo/esp32/modServo.c: In function 'xs_servo':
/Users/satoshi/Projects/moddable/modules/pins/servo/esp32/modServo.c:51:18: error: 'ledc_timer_config_t' has no member named 'bit_num'
   51 |                 .bit_num = LEDC_TIMER_15_BIT,
      |                  ^~~~~~~
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/servo/modServo.c.o] Error 1
make: *** Waiting for unfinished jobs....
/var/folders/5w/vzfzs0rs18vbxbdkvpmqsyq00000gn/T//ccJ7JYV4.s: Assembler messages:
/var/folders/5w/vzfzs0rs18vbxbdkvpmqsyq00000gn/T//ccJ7JYV4.s:2627: Warning: setting incorrect section attributes for .rodata.mod.0
```

phoddie, 2023-04-13:
Another good find. Thank you.

It looks like the field name `bit_num` has been deprecated for some time in favor of `duty_resolution`. There is a helpful comment in the ESP-IDF 4 headers (but not ESP-IDF 5!):

```c
    union {
        ledc_timer_bit_t duty_resolution;  /*!< LEDC channel duty resolution */
        ledc_timer_bit_t bit_num __attribute__((deprecated)); /*!< Deprecated in ESP-IDF 3.0. This is an alias to 'duty_resolution' for backward compatibility with ESP-IDF 2.1 */
    };
```

With that change, the build completes successfully. Since the only change is the name of the field, it should continue to work as always

This change should be live tomorrow (Friday in California).

phoddie, 2023-06-27:
Closing as the fix is now live.

closed 2023-06-27 by phoddie