Moddable-OpenSource/moddable #1089: There is a problem with the Match function

salarizadi, 2023-04-14:
I don't know if it might be a bug or not, but I told you to let me know

Note: I'm still using the ESP IDF 4 version because I'm building a project that I unfortunately can't deal with ESP IDF 5 at the moment.

The problem with the Match string function is that when we try to check with Regex, it does not return an array and does not identify it correctly. I tested this in the Chrome engine, it worked correctly, and because Moddable uses ES6 and also modern JavaScript, this problem should be solved.

my Regex:
```
const
    regex_whitespace = "[\\x20\\t\\r\\n\\f]",
    regex_identifier = "(?:\\\\[\\da-fA-F]{1,6}" + regex_whitespace + "?|\\\\[^\\r\\n\\f]|[\\w-]|[^\0-\\x7f])+",
    regex_strip_tag    = /(<([^>]+)>)/ig,
    regex_attributes   = /\s([^'"/\s><]+?)[\s/>]|([^\s=]+)=\s?(".*?"|'.*?')/g,
    regex_bracket_attr = "\\[" + regex_whitespace + "*(" + regex_identifier + ")(?:" + regex_whitespace + "*([*^$|!~]?=)" + 
    regex_whitespace + "*(?:'((?:\\\\.|[^\\\\'])*)'|\"((?:\\\\.|[^\\\\\"])*)\"|(" + regex_identifier + "))|)" + regex_whitespace + "*\\]";
```

I wrote this model: ```"[test='hello world']".match(new RegExp(regex_bracket_attr))```

Moddable output : 
  ```
  [
      "[test='hello world']",
      "test='hello world'",
      null,null,null,null
  ]
  ```

Chrome output : 
  ```
   [
      "[test='hello world']",
      "test",
      "=",
      "hello world",
      null,
      null
   ]
  ```

Chrome shows the correct result

phoddie, 2023-04-14:
Thank you for the report! It does look like a conformance issue. I'm able to reproduce your results in Safari. We'll take investigate.

phoddie, 2023-04-14:
Because of the zero character in `[^\0-\\x7f]`, the set matched all characters instead of characters not between 0 and 127. We have a fix. That will roll out in our next XS update. If you'd like to try it before then, the change is in [xsre.c](https://github.com/Moddable-OpenSource/moddable/blob/d77e86bc04601770c2ee531f98f2a7474c03efc4/xs/sources/xsre.c#L631-L634). You just need to move one line, so that this....

```c
			*current0++ = begin;
			character = end;
		}
	}
```

...becomes this...

```c
			*current0++ = begin;
		}
		character = end;
```

Thanks again for noticing the problem and taking the time to report it.

salarizadi, 2023-04-14:
Thanks a lot, I started with String.replaceAll for now, but I'm waiting for the new version of xs. It is very good that you fix the problems so quickly🙌

salarizadi, 2023-04-14:
> Because of the zero character in `[^\0-\\x7f]`, the set matched all characters instead of characters not between 0 and 127. We have a fix. That will roll out in our next XS update. If you'd like to [...]

Thanks a lot, I started with String.replaceAll for now, but I'm waiting for the new version of xs. It is very good that you fix the problems so quickly🙌

phoddie, 2023-04-26:
The fix for this is live. Please confirm and close. Thank you!

closed 2023-04-26 by salarizadi