Moddable-OpenSource/moddable #1092: Outline module bug

salarizadi, 2023-04-15:
When I set Height and Width to zero, it crashes and turns on and off continuously, but if I set the number to 1 or higher, there is no problem, what is the reason?

```
Outline.RoundRectPath(
         0, // left
         0, // top
         0, // width
         0, // height
         0 // radius
 )
```

phoddie, 2023-04-15:
A quick test here didn't crash. I tried both the simulator and an ESP32. Is there more to your test case?

https://user-images.githubusercontent.com/6856458/232250581-abf7a1d5-8f64-4906-a1b6-a876a1f24691.mov

salarizadi, 2023-04-15:
> A quick test here didn't crash. I tried both the simulator and an ESP32. Is there more to your test case?

Test this :

```
    shape = Shape(null, {
                width: screen.width, height: screen.height,
                Behavior: class {
                    onCreate ( shape, delta ) {
                        let path = Outline.RoundRectPath(
                             0, 0, 0, 0, 0
                         )
       
                        shape.fillOutline = Outline.fill(path);
                        shape.strokeOutline = Outline.stroke(
                               path, 5, Outline.LINECAP_BUTT, Outline.LINEJOIN_MITER
                        );
                    }
                }
            })
```

phoddie, 2023-04-16:
Thank you for the updated script. That shows the problem. In this case, the path generated by RoundRectPath causes Outline.stroke to generate a path with zero points in an unexpected way. That leads to a crash in FT_Stroker_GetCounts. We'll investigate the right place to fix that.

FWW – here's a simplified test:

```js
import {Outline} from "commodetto/outline";

let path = Outline.RoundRectPath(0, 0, 0, 0, 0);
Outline.stroke(path, 5, Outline.LINECAP_BUTT, Outline.LINEJOIN_MITER);
```

salarizadi, 2023-04-16:
> Thank you for the updated script. That shows the problem. In this case, the path generated by RoundRectPath causes Outline.stroke to generate a path with zero points in an unexpected way. That leads [...]

It's very good that you were able to quickly understand what the problem is, but be sure to solve this problem from the ground up, maybe we want to change the size from 0 and this will cause the system to fail.

salarizadi, 2023-04-17:
> Thanks for understanding. In fact, there are ways to trigger this same crash with other kinds of paths, not just round rectangles. We are testing a fix. If that checks out, we'll post it later this week.

Very good, if you want, I can cooperate and fix more errors together :)

phoddie, 2023-04-21:
This bug has been fixed by [this commit](https://github.com/Moddable-OpenSource/moddable/commit/ae3765bd4ba9bbd5a43a98b0803fba4d1940d7de).

> if you want, I can cooperate and fix more errors together :)

Please!

phoddie, 2023-04-26:
Please validate the fix and close. Thank you!

closed 2023-04-26 by salarizadi