Moddable-OpenSource/moddable #1094: Program crashes during TLS connection when xsbug serial is not connected

meganetaaan, 2023-04-23:
**Build environment:** Linux
**Target device:** `esp32/m5stack` and `pico/pico_w`
**Target Moddable version:** Latest release (https://github.com/Moddable-OpenSource/moddable/releases/tag/3.8.7)

**Description**
The program crashes during TLS connection only when the `xsbug` serial is not connected.

**Steps to Reproduce**
1. Build and install the `examples/io/tcp/httpsclient` using the following command:
```
mcconfig -d -m -p pico/pico_w ssid=my-ssid password=my-password
```
2. The program runs successfully with an xsbug connection.
3. Close xsbug.
4. From the next startup, the program repeatedly crashes and restarts.

**Expected behavior**
The TLS connection should work without an xsbug connection.

**Additional information**
- The `examples/io/tcp/httpclient` works correctly, suggesting that the issue lies within the TLSSocket.
- I have observed this behavior on two different Wi-Fi networks.

meganetaaan, 2023-04-23:
Solved with the help of @stc1988 . The root cause of the crash was the lack of NTP date sync which is done by the initialization by xsbug. After setting `config.sntp` in manifest config the date is synced and the TLSSocket worked successfully.

```manifest.json
"config": {
  "sntp": "pool.ntp.org"
}
```

phoddie, 2023-04-23:
Thank you to @stc1988 for helping to track this down.

TLS requires the correct time to be set to validate the date range on certificates. When debugging, xsbug automatically sets the current time on the MCU at start-up. This is fast and convenient. When xsbug is not connected, the MCU must set it. There are multiple approaches to this -- from network time via NTP, a local RTC clock component, etc.

All the examples that use TLS specify an NTP server so that the time is initialized immediately after Wi-Fi is connected. Is there something more that we should do to help others avoid this confusion in the future?

phoddie, 2023-05-06:
FYI – I made some changes to try to help other developers with their instrumented & release builds in the future.

I borrowed a [behavior](https://github.com/phoddie/node-red-mcu/blob/93de6348bea78d93b4c6d2d2e97d78cda0256944/setupwifi.js#L46) I implemented for Node-RED MCU Edition. In manifest_net.json there is now a default SNTP server defined. When Wi-Fi connects, if the MCU does **not** have a valid time, then the time is synchronized using over the network. If a valid time is set -- for example by xsbug or a battery-backed RTC -- then the synchronization with SNTP is not performed. This only does a time sync when required, so start-up of debugging builds remains fast. Projects can override this behavior to implement any policy they like.

These changes are for all ports that support Wi-Fi --ESP8266, ESP32, and Pico. They'll be available sometime next week.

closed 2023-05-06 by phoddie