Moddable-OpenSource/moddable #1102: Installing mod.### fxCommandReceived: remote operation id 0xff0f failed with resultCode -8

linfan68, 2023-05-07:
I'm trying to get a ModHost + Mod example working, based on this document:
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/mods.md

# Host
manifest
```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_typings.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODDABLE)/modules/base/modules/manifest.json"
	],
	"defines": {
		"XS_MODS": 1
	},
	"modules": {
		"*": [
			"./src/main"
		]
	}
}

```
main.ts
```ts
import Timer from 'timer'
import Modules from "modules"

trace("=============Hello World=============!\n")

if (Modules.has("rtkClient")) {
  trace("rtkClient module is available\n")
} else {
  trace("rtkClient module is not available\n")
}

let count = 0
Timer.repeat(() => {
  trace(`Second Tick ${count} \n`)
  count++
}, 1000)

```

# Mod
manifest:
```
{
  "include": [
    "$(MODDABLE)/examples/manifest_mod.json"
  ],
  "config": {
    "version": 1.1
  },
  "modules": {
    "*": [
      "./src/rtkClient"
    ]
  }
}
```

rtkClient.ts
```
# empty
```

After install host using 
```
mcconfig -d -m -p esp32/esp32s3
```
I run this from the mod project folder 
```
mcrun -d -m -p esp32/esp32s3
```

But got this output:
```
(tile) C:\DK\DKJS\jstest\rtkClient>mcrun -d -m -p esp32/esp32s3
# tsc tsconfig.json
# xsc rtkClient.xsb
# xsl rtkClient.xsa
Starting xsbug... 
"Running serial2xsbug..." 
Installing mod.### fxCommandReceived: remote operation id 0xff0f failed with resultCode -8
NMAKE : fatal error U1077: 'echo "Running serial2xsbug..." && set "XSBUG_PORT=5002" && set "XSBUG_HOST=localhost" && C:\DK\DKJS\moddable\build\bin\win\release\serial2xsbug COM4 460800 8N1 -install C:\DK\DKJS\moddable\build\bin\esp32\debug\rtkClient\rtkClient.xsa' : return code '0xffffffff'
Stop.
```

andycarle, 2023-05-07:
Try cleaning out your esp32 output in build\bin\esp32 and build\tmp\esp32 and then rebuilding/deploying both the host and mod.

I've seen occasional issues where stale build output that did not have mods enabled can cause this problem.

linfan68, 2023-05-07:
> Try cleaning out your esp32 output in build\bin\esp32 and build\tmp\esp32 and then rebuilding/deploying both the host and mod.
> 
> I've seen occasional issues where stale build output that did not have mods enabled can cause this problem.

It works!
I guess it's the new DEFINE needs a global recompile

closed 2023-05-07 by linfan68

andycarle, 2023-05-07:
Great! Glad to hear it's working for you.

There's a real bug here somewhere. That clean shouldn't be necessary.