Moddable-OpenSource/moddable #1103: Mod failed to load which more than 19 variables?

linfan68, 2023-05-07:
I'm trying to write a Mod "rtkClient"

in the Host side, the following code is used to load import the module
```
if (Modules.has("rtkClient")) {
  trace("rtkClient module is available\n")
  let rtkClient = Modules.importNow("rtkClient")
  trace(`rtkClient version: ${(rtkClient as any).version}\n`)
} else {
  trace("rtkClient module is not available\n")
}
```

and the normal output should be:
```
=============Hello World=============!
rtkClient module is available
=============RTK Client=============!
rtkClient version: undefined // this is OK
```

however, when the code in rtkClient gets a little bit complicated, the `Module.has` call failed like this:
```
=============Hello World=============!
rtkClient module is not available
......................................................................Mod failed: not enough keys
```

And I can reproduce this with this code:
```js
trace('=============RTK Client=============!\n')

const VAR_00 = 'var'
const VAR_01 = 'var'
const VAR_02 = 'var'
const VAR_03 = 'var'
const VAR_04 = 'var'
const VAR_05 = 'var'
const VAR_06 = 'var'
const VAR_07 = 'var'
const VAR_08 = 'var'
const VAR_09 = 'var'
const VAR_10 = 'var'
const VAR_11 = 'var'
const VAR_12 = 'var'
const VAR_13 = 'var'
const VAR_14 = 'var'
const VAR_15 = 'var'
const VAR_16 = 'var'
const VAR_17 = 'var'
const VAR_18 = 'var'

const VAR_19 = 'var' // <<=== This line makes module load failed
```

when I add `const VAR_19`, the module failed.

based on the "not enough keys" message, I tried to add this into my manifest(both host and mod)
```json
  "keys": {
    "available": 256,
    "name": 53,
    "symbol": 3
   }
```
but got not luck...

linfan68, 2023-05-07:
Figured it out
It IS the key issue, based on the document 
https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/mods.md#keys

> The dynamic nature of the JavaScript language means that the JavaScript engine needs to keep track of all properties names used by the virtual machine. XS does this using a key array. The [preload](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/preload.md) feature used for building hosts for microcontrollers stores all the host's property names into a key array in flash storage.
> At launch, XS creates a second array to store keys created at runtime. The length of this array is fixed when it is created. The manifest_base.json manifest file used by many projects configures the runtime key array to hold 32 keys. Because hosts usually create just a handful of keys at runtime, 32 is sufficient.
> Mods, however, may create many more symbols because they can contain code with many property names that do not appear in the host. To accommodate this, a project can adjust the length of the runtime key array by changing the number of available keys in the defines section of the manifest.

And adding this to the host's manifest file fixed the issue:
```
"creation": {
	"keys": {
		"available": 128
	}
}
```

closed 2023-05-07 by linfan68

phoddie, 2023-05-07:
Hello -- yes, there can be a limit on the number of keys available.

May I ask what version of the Moddable SDK you are running? In more recent versions (starting with 3.8.0, I believe) the key array should grow automatically, so then you don't need to manually adjust the creation parameters.

phoddie, 2023-05-08:
Updating the creation parameters in `$MODDABLE/examples/manfest_base.json`to include a non-zero value for `incremental` allows keys to grow automatically:

```
	"creation": {
		"keys": {
			"initial": 32,
			"incremental": 32,
			"name": 53,
			"symbol": 3
		},
```

This capability is documented in [manifest.md](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/manifest.md#keys).

I tested with the following mod that defines 30 unique symbols in the source, and another 200 when it runs.

<details>
<summary>test case</summary>

```js
let x00 = 0;
let x01 = 0;
let x02 = 0;
let x03 = 0;
let x04 = 0;
let x05 = 0;
let x06 = 0;
let x07 = 0;
let x08 = 0;
let x09 = 0;
let x10 = 0;
let x11 = 0;
let x12 = 0;
let x13 = 0;
let x14 = 0;
let x15 = 0;
let x16 = 0;
let x17 = 0;
let x18 = 0;
let x19 = 0;
let x20 = 0;
let x21 = 0;
let x22 = 0;
let x23 = 0;
let x24 = 0;
let x25 = 0;
let x26 = 0;
let x27 = 0;
let x28 = 0;
let x29 = 0;

const s = {}
for (let i = 0; i < 200; i++)
	s["t" + i] = i;
```

</details>

We'll update manifest_base.json to enable this by default.