Moddable-OpenSource/moddable #1118: Cannot connect in a wifi with 32 character

brenocastrocardoso, 2023-05-23:
**Build environment:** macOS
**Target device:**  ESP32

**Description**
The device cannot connect to an wifi with 32 character SSID.

**Steps to Reproduce**
1. Set up a wifi with name "wifitest_with32character_1234567"
2. Try to connect the application to that wifi
3. The exception happens "ssid too long - 31 bytes max" 

[image]

**Expected behavior**
The esp32 should be able to connect to a 32 bytes ssid as the standard define

**Other information**
There are some discussions on the esp-idf on this topic ([here](https://github.com/espressif/esp-idf/issues/3762)) seems like it was fixed and the struct dont expect a null terminated string anymore.

I tested removing this -1 to throw the exception only when the ssid is higher than 32 char and it worked as expected

[image]

phoddie, 2023-05-23:
You appear to be referencing out of date code. Here's the current implementation:

https://github.com/Moddable-OpenSource/moddable/blob/2420f19c2d3bda16104dcda7c18a8b94d3405436/modules/network/wifi/esp32/xs6wifi.c#L171-L186

brenocastrocardoso, 2023-05-24:
Oh, that explain. I tested with the updated code and it is working fine, thanks.

closed 2023-05-24 by brenocastrocardoso