Moddable-OpenSource/moddable #1119: esp32: support setting the attenuation in embedded:io/Analog

tve, 2023-05-24:
**Describe**
The esp32's analog inputs offer a selection of 4 attenuations, which correspond to different input voltage ranges. The resolution of the ranges varies in consequence. Please provide a option in the embeded:io/Analog constructor to choose the attenuation.

closed 2023-05-24 by tve

phoddie, 2023-05-24:
I see you closed this before I had a chance to read it.

If you'd like to revisit it, there are a couple of question:
- How to specify that in the constructor's options object? That could be done in a way that is specific to ESP32 or something more general that could be applicable to other silicon targets.
- The attenuation is not unique to each pin, but to groups of pins through the ESP32's ADC units. Would the first pin get to specify the attenuation for that group (and any mismatch would throw), or some other behavior?

tve, 2023-05-25:
Sorry for being unfriendly and just closing... When I realized the complications this would introduce I thought that the benefits don't outweigh the disadvantages. Unless the newer esp32's have vastly improved ADCs I think that the increased resolution of the lower ranges is just not worth it. Basically I thought that unless I can document that the 3 other attenuations provide a real benefit I should shut up :-).

phoddie, 2023-05-25:
Understood and agreed. The code to enable this capability is trivial, so it is mostly a design exercise if/when it would be useful to have.

FWIW – The ADCs on the ESP32 are fine for basic uses, but not spectacular For our commercial projects where accuracy really matters, we've used an external ADC peripheral connected via I²C instead.