Moddable-OpenSource/moddable #1120: embedded:io/Analog calibrates multiple times

tve, 2023-05-24:
**Build environment:** Linux
**Target device:** esp32-c3
**Description**
In embedded:io/Anaglog the calibration is called every time a channel is constructed as opposed to once per ADC unit as the code intends to do. The issue is that it checks the calibration structure's vref which remains 0 if a two-point calibration is performed. See: https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/analog/esp32/_analog.c#L165

**Steps to Reproduce**
1. Add some xsTrace printing to https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/analog/esp32/_analog.c#L166-L167
2. Create two Analog objects on the same ADC, e.g. using pins 0 and 1
3. Observe that the xsTrace printout occurs twice

I fixed the issue by checking for `gADC1Characteristics.adc_num` which gets set to 1 or 2 for the two ADCs. I'm not sure there is a big problem with calling the calibration for every object construction but the ESP-IDF docs do state that calibration must occur before reading values.

For reference, my quick hack:
```
@@ -162,10 +162,13 @@ void xs_analog_constructor_(xsMachine *the)
        if (kIOFormatNumber != builtinInitializeFormat(the, kIOFormatNumber))
                xsRangeError("invalid format");
 
-       if (port == 1 && !gADC1Characteristics.vref) {
+       if (port == 1 && !gADC1Characteristics.adc_num) {
                adc1_config_width(ADC_WIDTH);
-               esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN, ADC_WIDTH, V_REF, &gADC1Characteristics);
-       } else if (port == 2 && !gADC2Characteristics.vref) {
+               esp_adc_cal_value_t v = esp_adc_cal_characterize(ADC_UNIT_1, ADC_ATTEN, ADC_WIDTH, 2000, &gADC1Characteristics);
+               char msg[32];
+               sprintf(msg, "cal:%d adc:%d\n", v, gADC1Characteristics.adc_num);
+               xsTrace(msg);
+       } else if (port == 2 && !gADC2Characteristics.adc_num) {
                esp_adc_cal_characterize(ADC_UNIT_2, ADC_ATTEN, ADC_WIDTH, V_REF, &gADC2Characteristics);
        }
```
the value printed for `v` is 1 which corresponds to `ESP_ADC_CAL_VAL_EFUSE_TP`

phoddie, 2023-05-24:
Thanks for the report and proposed fix. It makes sense to not use vref to check that the ADC has been calibrated.

The change depends on ADC_UNIT_1 and ADC_UNIT_2 having non-zero values (and these are unusual as many ESP-IDF enums start at 0 instead of 1). Do you have any concerns with doing this instead?

```c
static esp_adc_cal_characteristics_t gADC1Characteristics = {.adc_num = ADC_UNIT_MAX + 1}, gADC2Characteristics = {.adc_num = ADC_UNIT_MAX + 1};
...
	if (port == 1 && (gADC1Characteristics.adc_num > ADC_UNIT_MAX)) {
...
```

----

FWIW – instead of this...

```c
char msg[32];
sprintf(msg, "cal:%d adc:%d\n", v, gADC1Characteristics.adc_num);
xsTrace(msg);
```

...you can write this:

```c
xsLog("cal:%d adc:%d\n", (int)v, (int)gADC1Characteristics.adc_num);
```

Smaller, faster, and no risk of buffer overflow.

tve, 2023-05-25:
I used adc_num as-is 'cause I've not seen an esp32 with an ADC0, but your option sounds good too.

Thanks for the heads-up about xsLog, I would have used that if it was mentioned in https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md ;-)

phoddie, 2023-05-25:
Thanks for the feedback. For our next SDK update, I've modified the ESP32 analog input implementation as discussed and added `xsLog()` to the XS in C documentation.

closed 2023-06-05 by phoddie

tve, 2023-06-06:
> and added xsLog() to the XS in C documentation.

Thanks for adding, could you also add which `%` formats are supported? I just tried `%f` and that doesn't work. Looks like the formatting happens in the debugger, that makes the set of supported formats non-obvious to discover...

phoddie, 2023-06-06:
[FYI](https://github.com/Moddable-OpenSource/moddable/blob/24d84d7b340c6bfda73b3b1499ebf88d59aaeec8/xs/sources/xsDebug.c#L993-L1033).