Moddable-OpenSource/moddable #1123: XS returns the wrong result for expressions that assign to array length

gibson042, 2023-05-26:
**Environment:** XS 13.0.0 32 4

**Description**
Evaluating `array.length = nonNumber` does not conform with the ECMAScript specification.

**Steps to Reproduce**
```js
typeof ([].length = "4")
```

**Actual behavior**
"number"

**Expected behavior**
"string"

**Other information**
[Evaluation of ```AssignmentExpression : LeftHandSideExpression `=` AssignmentExpression```](https://tc39.es/ecma262/multipage/ecmascript-language-expressions.html#sec-assignment-operators-runtime-semantics-evaluation) when |LeftHandSideExpression| is not an object or array literal and the right-hand-side |AssignmentExpression| is not a function definition is specified in step 1.e to return the result of evaluating that |AssignmentExpression| in step 1.c.

Basically, even though the intervening [PutValue](https://tc39.es/ecma262/multipage/ecmascript-data-types-and-values.html#sec-putvalue) can perform arbitrary processing on that value (as is the case when setting the length of an array ends up in [ArraySetLength](https://tc39.es/ecma262/multipage/ordinary-and-exotic-objects-behaviours.html#sec-arraysetlength) and the value is coerced to a uint32 number), the assignment expression itself must evaluate to the input of that process rather than its output.

Note also that this affects all assignment operators, e.g. `Array(1).length += "e2"` should be the string "1e2".

phoddie, 2023-05-26:
Thanks for the report.

XS is very deliberate about returning a number from the Array length setter. Here's the relevant code:

https://github.com/Moddable-OpenSource/moddable/blob/2420f19c2d3bda16104dcda7c18a8b94d3405436/xs/sources/xsArray.c#L996-L997

Changing that to return the input argument gives the expected result for the two test cases you provide:

```c
mxResult->value = mxArgv(0)->value;
mxResult->kind = mxArgv(0)->kind;
```

But, we'll need to take a more careful look.

phoddie, 2023-06-05:
Fixed

closed 2023-06-05 by phoddie