Moddable-OpenSource/moddable #1125: build  examples\network\ble\health-thermometer-server meet `libsodium` fatal error

rillhu, 2023-05-28:
Environment: 
moddable sdk 3.9.2
windows 11

Description
build `examples\network\ble\health-thermometer-server`, and get below error message:
```
[98/98] cmd.exe /C "cd /D C:\Users\admin\Projects\moddable\build\tmp\esp32\wrover_kit\debug\health-thermometer-server\xsProj-esp32\build\bootloader\esp-idf\esptool_py && C:\Users\admin\.espressif\python_env\idf4.4_py3.11_env\Scripts\python.exe C:/esp/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/Users/admin/Projects/moddable/build/tmp/esp32/wrover_kit/debug/health-thermometer-server/xsProj-esp32/build/bootloader/bootloader.bin"
Bootloader binary size 0x4420 bytes. 0x2be0 bytes (39%) free.
[970/1148] Building C object esp-idf/libsodium/CMakeFiles/...ult/ristretto255/ref10/scalarmult_ristretto255_ref10.c.obj
FAILED: esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_scalarmult/ristretto255/ref10/scalarmult_ristretto255_ref10.c.obj
ccache C:\Users\admin\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DCONFIGURED -DHAVE_WEAK_SYMBOLS -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DNATIVE_LITTLE_ENDIAN -D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -IC:/Users/admin/Projects/moddable/build/tmp/esp32/wrover_kit/debug/health-thermometer-server/xsProj-esp32/build/config -IC:/esp/components/libsodium/libsodium/src/libsodium/include -IC:/esp/components/libsodium/port_include -IC:/esp/components/libsodium/libsodium/src/libsodium/include/sodium -IC:/esp/components/libsodium/port_include/sodium -IC:/esp/components/libsodium/port -IC:/esp/components/newlib/platform_include -IC:/esp/components/freertos/include -IC:/esp/components/freertos/include/esp_additions/freertos -IC:/esp/components/freertos/port/xtensa/include -IC:/esp/components/freertos/include/esp_additions -IC:/esp/components/esp_hw_support/include -IC:/esp/components/esp_hw_support/include/soc -IC:/esp/components/esp_hw_support/include/soc/esp32 -IC:/esp/components/esp_hw_support/port/esp32/. -IC:/esp/components/esp_hw_support/port/esp32/private_include -IC:/esp/components/heap/include -IC:/esp/components/log/include -IC:/esp/components/lwip/include/apps -IC:/esp/components/lwip/include/apps/sntp -IC:/esp/components/lwip/lwip/src/include -IC:/esp/components/lwip/port/esp32/include -IC:/esp/components/lwip/port/esp32/include/arch -IC:/esp/components/soc/include -IC:/esp/components/soc/esp32/. -IC:/esp/components/soc/esp32/include -IC:/esp/components/hal/esp32/include -IC:/esp/components/hal/include -IC:/esp/components/hal/platform_port/include -IC:/esp/components/esp_rom/include -IC:/esp/components/esp_rom/include/esp32 -IC:/esp/components/esp_rom/esp32 -IC:/esp/components/esp_common/include -IC:/esp/components/esp_system/include -IC:/esp/components/esp_system/port/soc -IC:/esp/components/esp_system/port/public_compat -IC:/esp/components/esp32/include -IC:/esp/components/xtensa/include -IC:/esp/components/xtensa/esp32/include -IC:/esp/components/driver/include -IC:/esp/components/driver/esp32/include -IC:/esp/components/esp_pm/include -IC:/esp/components/esp_ringbuf/include -IC:/esp/components/efuse/include -IC:/esp/components/efuse/esp32/include -IC:/esp/components/vfs/include -IC:/esp/components/esp_wifi/include -IC:/esp/components/esp_event/include -IC:/esp/components/esp_netif/include -IC:/esp/components/esp_eth/include -IC:/esp/components/tcpip_adapter/include -IC:/esp/components/esp_phy/include -IC:/esp/components/esp_phy/esp32/include -IC:/esp/components/esp_ipc/include -IC:/esp/components/app_trace/include -IC:/esp/components/esp_timer/include -IC:/esp/components/mbedtls/port/include -IC:/esp/components/mbedtls/mbedtls/include -IC:/esp/components/mbedtls/esp_crt_bundle/include -mlongcalls -Wno-frame-address  -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Os -freorder-blocks -fmacro-prefix-map=C:/Users/admin/Projects/moddable/build/tmp/esp32/wrover_kit/debug/health-thermometer-server/xsProj-esp32=. -fmacro-prefix-map=C:/esp=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4.3-dirty\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -Wno-unused-function -MD -MT esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_scalarmult/ristretto255/ref10/scalarmult_ristretto255_ref10.c.obj -MF esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_scalarmult\ristretto255\ref10\scalarmult_ristretto255_ref10.c.obj.d -o esp-idf/libsodium/CMakeFiles/__idf_libsodium.dir/libsodium/src/libsodium/crypto_scalarmult/ristretto255/ref10/scalarmult_ristretto255_ref10.c.obj -c C:/esp/components/libsodium/libsodium/src/libsodium/crypto_scalarmult/ristretto255/ref10/scalarmult_ristretto255_ref10.c
C:/esp/components/libsodium/libsodium/src/libsodium/crypto_scalarmult/ristretto255/ref10/scalarmult_ristretto255_ref10.c:63:1: fatal error: opening dependency file esp-idf\libsodium\CMakeFiles\__idf_libsodium.dir\libsodium\src\libsodium\crypto_scalarmult\ristretto255\ref10\scalarmult_ristretto255_ref10.c.obj.d: No such file or directory
 }
 ^
compilation terminated.
[979/1148] Building C object esp-idf/freemodbus/CMakeFiles/__idf_freemodbus.dir/modbus/functions/mbfuncholding.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

Steps to Reproduce
cd %moddable%\examples\network\ble\health-thermometer-server
`mcconfig -d -m -p esp32/wrover_kit`

Expected behavior
build successful

Other information

phoddie, 2023-05-28:
That compilation error is coming from the ESP-IDF build. I just build the example here using macOS and there were no errors.

Some questions to try to narrow it down:

- Are you using the recommended ESP-IDF version. In my build, the Moddable SDK outputs `Using recommended ESP-IDF v4.4.3` to the console. Do you see that?
- It could be a problem with parallel builds on Windows. If you simply build again, does it change?
- Do other Moddable examples build for you?

andycarle, 2023-05-28:
This error is usually related to hitting the maximum path length on Windows during the ESP-IDF build. It's a little silly, but one reliable fix is to move your `moddable` directory to someplace with a shorter path (like just `C:\moddable`).

rillhu, 2023-05-29:
thanks @phoddie and @andycarle, 
Firstly, yes, I use the officail esp-idf release v4.4.3. And it works for some other example such as `hello world` . Regarding this example, I tried to build it for several times, the issue cannot be solved. 

Then I tried the advice of Andy (move moddable to C:\moddable), it works.
Thanks you all for so fast help here.