Moddable-OpenSource/moddable #1128: freertos/FreeRTOS.h not found IDF-5

salarizadi, 2023-06-01:
I am working on an advanced but small bot when it was built with IDF 4 it had no problems but when the update was done!

I install new version of moddableSDK IDF-5 from : https://github.com/Moddable-OpenSource/moddable/tree/idf-v5

When run this command for build : mcconfig -m -p esp32/nodemcu -f rgb565le

#include "freertos/FreeRTOS.h"  Not Found Why?

```
In file included from C:\Espressif\frameworks\esp-idf-v5.0.2\components\lwip\port\esp32\include/lwipopts.h:23,
                 from C:\Espressif\frameworks\esp-idf-v5.0.2\components\lwip\lwip\src\include/lwip/opt.h:51,
                 from C:\Espressif\frameworks\esp-idf-v5.0.2\components\lwip\lwip\src\include/lwip/netdb.h:37,
                 from D:\ModdableSDK\xs\platforms\esp/xsPlatform.h:113,
                 from D:\ModdableSDK\xs\sources/xsCommon.h:41,
                 from D:\ModdableSDK\xs\sources/xsAll.h:41,
                 from D:\ModdableSDK\xs\platforms\esp\xsHost.c:38:
C:\Espressif\frameworks\esp-idf-v5.0.2\components\esp_system\include/esp_task.h:24:10: fatal error: freertos/FreeRTOS.h: No such file or directory
   24 | #include "freertos/FreeRTOS.h"
      |          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.
NMAKE : fatal error U1077: 'C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.EXE' : return code '0x1'
Stop.
```

phoddie, 2023-06-01:
Thanks for trying out our ESP-IDF 5 preview! This problem hasn't come up before.

The include that is failing is coming from source file inside the ESP-IDF itself  (where esp_task.h is including freertos/FreeRTOS.h). We'll take a look to see if we can reproduce this. 

FWIW – I usually work on macOS and it looks like your error report is from Windows. One possibility is that this only happens on Windows.  That said, I know @stc1988 did quite a bit of testing of our ESP-IDF 5 preview. I believe he may have been working on Windows.

stc1988, 2023-06-02:
I tried out IDF v5.0.1 on macOS.
But I can try this on Windows.

mkellner, 2023-06-02:
Hello. The initial work was done on MacOS. The windows nmake.esp32.mk is not updated for v5.x. I am looking at this now.

ufocms, 2023-06-02:
it's interesting! But I don't think it has anything to do with the operating system because the exact same FreeRtos.h file is not available in the include ESPIDF version 5 folder. I don't have a Mac system to test it and it was only tested with Windows, but I still don't think it has anything to do with the operating system.

salarizadi, 2023-06-02:
it's interesting! But I don't think it has anything to do with the operating system because the exact same FreeRtos.h file is not available in the include ESPIDF version 5 folder. I don't have a Mac system to test it and it was only tested with Windows, but I still don't think it has anything to do with the operating system.

mkellner, 2023-06-02:
The macOS and Linux builds use `tools/mcconfig/make.esp32.mk` as a template for the project build files. It needed to be changed for for IDF-v5 support because various files moved.

Those changes need to be applied to `tools\mcconfig\nmake.esp32.mk` which is a template for the project build files created by `mcconfig` on Windows hosts.

salarizadi, 2023-06-02:
I was not informed! So it makes sense that I have encountered this problem

phoddie, 2023-06-02:
@stc1988 Thank you. Once @mkellner has this updates, it would be very helpful if you could also try on Windows. There are always variations in machine configuration, so more people testing makes a difference.

mkellner, 2023-06-02:
I have updated the `idf-v5` branch.
I have tried with ESP-IDF v5.0 on macOS and Windows 10.

salarizadi, 2023-06-03:
> I have updated the `idf-v5` branch. I have tried with ESP-IDF v5.0 on macOS and Windows 10.

This is great, I will check it soon and let you know.

phoddie, 2023-06-27:
Closing as this should now be resolved. Thanks for the report!

closed 2023-06-27 by phoddie