Moddable-OpenSource/moddable #1129: Unable to simply toggle an LED on pico

bartmichu, 2023-06-01:
I've noticed that simple LED toggle is not possible on pico when using the IO module with this pattern: ```led.write(!led.read())```

**Code example**
```javascript
import Digital from 'embedded:io/digital';

const led = new Digital({
  pin: 25,
  mode: Digital.Output
});

const button = new Digital({
  pin: 22,
  mode: Digital.Input,
  edge: Digital.Falling,
  onReadable () {
    led.write(!led.read());
  }
});
```

**More info**
Example code fails due to these checks:

```c
if (!digital->isInput)
		xsUnknownError("unimplemented");
```

Is it intentional? I'm a bit surprised as it used to work well on ESP platforms the last time I was playing with Moddable.

phoddie, 2023-06-01:
You are calling `read` on the LED. The LED is an output, so it is write-only. The check you indicate above is trying to tell you that:

https://github.com/Moddable-OpenSource/moddable/blob/2420f19c2d3bda16104dcda7c18a8b94d3405436/modules/io/digital/pico/digitalbank.c#L319-L325

The ESP32 implementation has an [identical check](https://github.com/Moddable-OpenSource/moddable/blob/2420f19c2d3bda16104dcda7c18a8b94d3405436/modules/io/digital/esp32/digitalbank.c#L297-L303).

Are you certain this exact code worked as-is on ESP32?

bartmichu, 2023-06-01:
According to my git commits I was using this on NodeMCU and Moddable One about two years ago. I'll try to dig out my Moddable One board and test it properly - as I should've done before opening this issue, to be honest.

bartmichu, 2023-06-01:
OK @phoddie so I have double checked now and this exact code works as I expected on ESP8266 (NodeMCU).

phoddie, 2023-06-01:
Thanks for the additional details. In the original report you wrote "ESP platforms." I interpreted that as ESP32, but actually it was ambiguous. ;) ESP8266 does **not** perform the check, so that explains why you didn't get an exception. Interesting that read returns a useful value for you there. Still, I'm sure that's not a reliable behavior across devices.

Embarrassingly, I left myself a note to add the error check in the ESP8266 implementation:

https://github.com/Moddable-OpenSource/moddable/blob/2420f19c2d3bda16104dcda7c18a8b94d3405436/modules/io/digital/esp/digitalbank.c#L437-L440

I'll fill that in now so there is a consistent behavior across silicon architectures.

Thank you for the report!

bartmichu, 2023-06-04:
Could you please clarify one thing for me.

I was not sure about the meaning of "unimplemented", but now when you have changed the message, am I right assuming that it will not be implemented at all?

Is it something unique to the IO module? Or a bad idea in general - the traditional `pins/digital` module allows reading an output as well - and you would advise not using it like that?