Moddable-OpenSource/moddable #1135: main.js (8) # Break: repeat: C: xsArg(1): invalid index!

salarizadi, 2023-06-03:
Hi.

In the previous version of Moddable, when the argument was not defined, no error occurred, but now it gives an error.

See the following example :

`
Timer.repeat(function () {
    trace(`Hello world`)
})
`

The time parameter is not defined for it and it gives an error, but this did not happen in version 3.5 of the ModdableSDK that I was working on.

phoddie, 2023-06-03:
Curious. The second argument is required, so it is appropriate that it is throwing now. Looking at the sources of v3.5, the first line of the native `repeat` implementation accesses the second argument (`xsArg(1)`). That will throw an exception if the argument is not present.

https://github.com/Moddable-OpenSource/moddable/blob/f9a07875de9e1e0e52e10e1b139d9d980d6c118b/modules/base/timer/modTimer.c#L106-L111

salarizadi, 2023-06-03:
> 

This was an example. I am building a new module as you know. When I want to get xsArg(2) but this argument can be defined or empty, even when I identify xsUndefinedType, the same error still occurs.

Note: Unfortunately, I thought this was not a problem in version 3.5, but now that I have returned to this version, I still could not solve this problem, it gives xsArg(2) an error.

salarizadi, 2023-06-03:
C :
`
void xs_write ( xsMachine *the ) {
    const uint8_t address = xsmcToInteger(xsArg(0)), value = xsmcToInteger(xsArg(1));
    const uint8_t n = xsmcToInteger(xsArg(2));

   // Other code....

    xsmcSetInteger(xsResult, n);
}
`

JS : 
`
class {
   write ( address, value, n ) @ "xs_write"
}
`

salarizadi, 2023-06-03:
@phoddie What do you think is the problem?

phoddie, 2023-06-03:
> When I want to get xsArg(2) but this argument can be defined or empty, even when I identify xsUndefinedType, the same error still occurs.

Are you asking about how to check for the presence of an argument using XS in C? 

In JavaScript this is done with `arguments.length `. In [XS in C](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md), you get the number of arguments using `xsToInteger(xsArgc))` or `xsmcArgc`.

salarizadi, 2023-06-03:
> 

In this code example, the xsUndefinedType check is not written, but I tested it a few hours ago and it gave an error again.

https://github.com/Moddable-OpenSource/moddable/issues/1135#issuecomment-1575092543

I don't know where the mistake is, can you send a sample code?

Thankful.

salarizadi, 2023-06-03:
@phoddie
@mkellner 

You are really great. Thanks a lot, problem solved, I thought I should check with xsUndefinedType, I didn't know that.

I have one more question for you. Why was I not confirmed in the list of participants after 4 weeks of signing the contract?

phoddie, 2023-06-05:
> Thanks a lot, problem solved, I thought I should check with xsUndefinedType, I didn't know that.

Glad that's resolved!

FWIW – the final chapter in [our book](https://www.moddable.com/book) is all about interfacing C code to JavaScript. It covers the essentials and includes working examples to show how the pieces work together. For developers getting started with XS in C, it is a great resource.

> Why was I not confirmed in the list of participants after 4 weeks of signing the contract?

Thank you for completing the Contributor License Agreement (CLA). The list of contributors shown on our main GitHub page is generated automatically by GitHub. Moddable doesn't control who is listed there.

closed 2023-06-05 by phoddie