Moddable-OpenSource/moddable #1137: mcmanifest munges typescript module specifiers

tve, 2023-06-04:
**Build environment:** Linux
**Moddable SDK version:** 3.9.2
**Target device:** lin
**Description**
I have a library module written in typescript for which I'm trying to use an import specifier of "embedded:lib/pid" and mcconfig, when it produces the `tsconfig.json`, munges this into `~.embedded/lib/pid`

**Steps to Reproduce**
1. unpack the attached tgz which contains a main that imports `embedded:lib/pid`
2. `run mcconfig -d -m -p lin -t build`
3. observe error:
```
main.ts:1:17 - error TS2792: Cannot find module 'embedded:lib/pid'. Did you mean to set the 'moduleResolution' option to 'nodenext', or to add aliases to the 'paths' option?

1 import PID from "embedded:lib/pid"
```
4. in `$(MODDABLE)/build/tmp/lin/mc/debug/app/modules/tsconfig.json` observe:
```
                        "~.embedded/lib/pid": [
                                "/home/src/moddable/repro-tsimport/modules/pid/pid"
                        ]
```

phoddie, 2023-06-04:
Please provide the manifest for your module as well. Thank you.

tve, 2023-06-04:
It's there:
```
> tar tfz repro.tsimport.tar.gz 
repro-tsimport/
repro-tsimport/modules/
repro-tsimport/modules/pid/
repro-tsimport/modules/pid/pid.ts
repro-tsimport/modules/pid/manifest.json  <==========
repro-tsimport/app/
repro-tsimport/app/doit
repro-tsimport/app/manifest.json
repro-tsimport/app/main.ts
repro-tsimport/tsconfig.json
```
or am I misunderstanding?

tve, 2023-06-04:
Argh, didn't realize I had to link to the attachment, my apologies. Here it is:

[repro.tsimport.tar.gz](https://github.com/Moddable-OpenSource/moddable/files/11646366/repro.tsimport.tar.gz)

tve, 2023-06-04:
I should add that the following hack makes things work as I would expect:
```
--- a/tools/mcmanifest.js
+++ b/tools/mcmanifest.js
@@ -1028,6 +1028,10 @@ export class TSConfigFile extends FILE {
                        var specifier = result.target.slice(0, -4);
                        if (tool.windows)
                                specifier = specifier.replaceAll("\\", "/");
+                       if (specifier.startsWith("~."))
+                               specifier = specifier.slice(2);
+                       specifier = specifier.replace("embedded/", "embedded:");
                        paths[specifier] = [ result.source.slice(0, -3) ];
                        json.files.push(result.source);
                }
```
but a real fix needs to address the issue earlier in the processing...

phoddie, 2023-06-05:
Thank you for the files. That's more clear now.

The "munging" is expected here. It is used to generate the file paths for build output of modules that use the "embedded" namespacee prefix because "embedded:whatever" doesn't work on all platforms for a directory name. Changing the extensions of main.ts and pid.ts to .js in your example allows everything to work as expected, even with the munging. The exception you see comes from the TypeScript compiler because the module specifier is incorrect in tsconfig.json, so it can't locate the "embedded:lib/pid" references in main.ts.

Your change to the code that generates the tsconfig.json is in the right place to fix that. A little more investigation is needed to understand if the change should be more general (e.g. support any namespace prefix, not just "embedded").

tve, 2023-06-05:
haha, hiding in plain sight! works indeed, thanks for the update!
dunno whether you want to close this issue or keep it open 'til the fix is released...

closed 2023-06-27 by phoddie