Moddable-OpenSource/moddable #1140: mcmanifest overlooks .d.ts files in modules with native code

tve, 2023-06-06:
**Build environment:** Linux
**Moddable SDK version:** 3.9.2
**Target device:** lin

**Description**
I'm writing a module with some native code and I need to import it in typescript, so I need a .d.ts file. But mcmanifest makes that difficult. Specifically, in the tsimport.json file it generates it needs to produce a `path` based on the .d.ts file and it doesn't

**Steps to Reproduce**
1. Extract the project from the attached [repro.tsnative.tar.gz](https://github.com/Moddable-OpenSource/moddable/files/11666923/repro.tsnative.tar.gz), change the `moddable` symlink to point to your `$MODDABLE` (the symlink is used in the tsconfig so one can run tsc manually or in vscode)
2. In the app dir, run `tsc` to verify that it's happy (it will show errors for a bunch of files in the moddable tree that would be good to fix, but that's a tangent and can be ignored here)
3. Try to build the project using `mcconfig -d -m -p lin -t build`
4. Observe the fatal error from tsc:
```
main.ts:1:18 - error TS2792: Cannot find module 'embedded:lib/hello'. Did you mean to set the 'moduleResolution' option to 'nodenext', or to add aliases to the 'paths' option?

1 import incr from "embedded:lib/hello"
```
5. Observe that no path for the `hello` module got included in the generated tsconfig

**Expected behavior**
I expected that mcconfig would have picked up the presence of `hello.d.ts` and added the appropriate path to the generated tsconfig

**Other information**
- In mcmanifest the `appendSource` method of `ModulesRule` is called for both `hello.c` and `hello.js` but not `hello.d.ts`, debug traces I added show
```
appendSource(~.embedded/lib/hello, /home/src/moddable/repro-tsnative/modules/hello/hello.c, .c)                                                                                 
appendSource(~.embedded/lib/hello, /home/src/moddable/repro-tsnative/modules/hello/hello.js, .js)                                                                               
```
- mcmanifest does recognize `.d.ts` files when using a wildcard in the manifest, so in the hello module's manifest, replacing the path in the modules section by ` "embedded:lib/*": "./*"` makes things work (but this shouldn't be needed and is definitely difficult to discover)
- it would be great if one could write the JS part of a native module in typescript directly! The issue, of course, are the `@` annotations. The simplest I can come up with is to move the annotations into comments and use the `removeComments: false` option in the tsconfig compilerOptions, then post-process the generated js files. A sample `hello.ts-` is included in the archive to illustrate how this might look (not all comments are preserved by tsc, see https://github.com/microsoft/TypeScript/issues/1665).

phoddie, 2023-06-06:
There are two topics here, as I understand it. The first concerns tools, mcconfig & mcrun. When building a JavaScript source file ("*.js"), an adjacent TypeScript declaration file (*.d.ts") is being ignored. That sounds like a bug and we should sort that out. The second concerns the syntax XS uses to bind native functions. Because it is not recognized by `tsc`, it is not possible to bind native functions from a TypeScript source file The second is a more complex topic that could be addressed in many different ways, so let's set that aside for this issue.

phoddie, 2023-06-06:
A quick patch to mcmanifest.js gets the .d.ts working. This isn't extensively tested, but please give it a try.

In `Rule.iterate`, after `splitPath`, check for the special case of the *.d.ts extension and fix up the result. (Maybe we should change splitPath to do this?)

```js
				var parts = tool.splitPath(path);
				if ((".ts" === parts.extension) && parts.name.endsWith(".d")) {
					parts.name = parts.name.slice(0, parts.name.length - 2);
					parts.extension = ".d.ts";
				} 
```

In `ModuleRules.appendSource` update the handling of the *.ts extension to handle the separate *.d.ts extension:

```js
		else if (parts.extension == ".ts")
			this.appendFile(tool.tsFiles, target + ".xsb", source, include);
		else if (parts.extension == ".d.ts")
			this.appendFile(tool.dtsFiles, target, source, include);
```

In `TSConfigFile.generate` there's no longer a ".d" to strip to generate the specifier:

```js
		for (var result of tool.dtsFiles) {
			var specifier = result.target;
```

tve, 2023-06-06:
I'll give this a spin...

closed 2023-06-27 by phoddie