Moddable-OpenSource/moddable #1145: JSON.stringify cannot render preloaded object

tve, 2023-06-08:
**Build environment:** Linux
**Moddable SDK version:** 3.9.2
**Target device:** Moddable Two

**Description**
Trying to JSON.stringify a preloaded object throws an exception:
```
var test = { x: 1, y: 2 }
trace(`test: ${JSON.stringify(test)}\n`)

export default function() {
    trace(`test: ${JSON.stringify(test)}\n`)
}
```
The manifest contains `"preload": "main"`.
The first trace is printed at compile time as expected.
The second fails with "stringify: read only value!".

**Expected behavior**
I expected JSON.stringify to succeed.

phoddie, 2023-06-08:
Yes, this has been reported before (see #340). There's no immediate intention to address it. A workaround is to use `structuredClone`.

closed 2023-06-08 by phoddie

tve, 2023-06-08:
thanks, my searches didn't show this one...
two suggestions: document it (e.g. in the XS preload doc in the section on what can't be preloaded) and detect it in JSON;.stringify and produce a more explicit error and perhaps even suggest structuredClone

phoddie, 2023-06-08:
I've updated the doc. Thank you for the suggestion.

Adding documentation into the runtime is a slippery slope, and not great for a runtime targeting extremely resource constrained targets.