Moddable-OpenSource/moddable #1149: poco truncates display list due to lack of memory

tve, 2023-06-10:
**Build environment:**  Linux
**Moddable SDK version:** 3.9.1
**Target device:** Moddable Two

**Description**
Depending on memory configuration poco silently stops drawing.

**Steps to Reproduce**
1. Extract  [repro-poco.tar.gz](https://github.com/Moddable-OpenSource/moddable/files/11713098/repro-poco.tar.gz)
2. Run in simulator as `mcconfig -v -d -m -p lin/moddable_two` (s/lin/mac/ as needed)
3. Observe a flattened cube in the display
4. In the manifest, change the chunk initial allocation to 2048 and rerun
5. Observe that only the red face of the cube gets drawn and that there is no error anywhere in sight
6. In the manifest, change the chunk incremental allocation to 4096 and rerun
7. Observe that now two faces of the cube are drawn

**Expected behavior**
Since the static allocation is unchanged in these 3 versions I expected the full flattened cube to be drawn even with a smaller initial allocation. Or, if there is an issue allocating a large contiguous chunk, that an error be reported with details so I can adjust the allocation parameters.

phoddie, 2023-06-10:
> 1. Extract [repro-poco.tar.gz](https://github.com/Moddable-OpenSource/moddable/files/11713098/repro-poco.tar.gz)
> 2. Run in simulator as mcconfig -v -d -m -p lin/moddable_two (s/lin/mac/ as needed)

This fails because of a missing module dependency:

```
### Error: '../modules/lib/fusion/manifest.json': manifest not found!
```

Removing  that line from the manifest results in this error:

```
main.ts:35:29 - error TS2554: Expected 2-5 arguments, but got 1.

35     const outline = Outline.stroke(path)
                               ~~~~~~~~~~~~

  ../../Projects/moddable/typings/commodetto/outline.d.ts:79:41
    79     static stroke(path: CommodettoPath, weight: number, linecap?: number, linejoin?: number, miterLimit?: number): Outline
                                               ~~~~~~~~~~~~~~
    An argument for 'weight' was not provided.

Found 1 error in main.ts:35
```

tve, 2023-06-10:
whoops! I must have fat-fingered the tgz upload

tve, 2023-06-10:
[repro-poco.tar.gz](https://github.com/Moddable-OpenSource/moddable/files/11713316/repro-poco.tar.gz)

phoddie, 2023-06-12:
Thanks, I'm able to reproduce this. Not sure what's going on yet. Have some other things to take care of today, so may not get to it right away.

phoddie, 2023-06-12:
If it was an allocation failure, it would report an error. That got me thinking about other failure modes which led me to understanding the problem. The outlines are being garbage collected before they can be rendered. That's XS being really efficient and trying not to expand the chunk heap unless necessary. That problem only happens when using Poco without Pit, as Piu manages the lifecycle of the outlines it creates. A clean, efficient solution will take a little effort to sort out.

tve, 2023-06-12:
I see, so something (probably poco) needs more memory, the heap is full, gc is triggered, it erroneously collects the outlines, interesting stuff ensues...?
And this is triggered by the fact that by the time poco.end() is called the variables holding the references to the outlines are long garbage and internally a gc-visible reference is not held?

phoddie, 2023-06-12:
Basically, yes. A simple workaround is to put the outlines in an Array that persists until `end` is called. More or less, that's probably the cleanest fix to this in Poco too.

phoddie, 2023-06-15:
Good news. I've got a solution working. The RAM overhead about as small as practical - 4 bytes per Poco instance plus (roughly) 4 bytes for each outline drawn between begin & end. That should roll out to this repository Friday.

closed 2023-06-27 by phoddie