Moddable-OpenSource/moddable #1158: Add schedule functionality to io/system

bartmichu, 2023-06-17:
Hello.

Looking at your examples I see that using `io/system` wrappers is a preferred
 way of working with timers when using other `io` modules. At the moment, schedule functionality is not implemented and I find this problematic.

I'm working with some lousy push buttons and need to implement throttling. The following code example works as intended (do something only when hold for two seconds):

```js
import Digital from 'embedded:io/digital';
import Timer from 'timer';

const holdTimer = Timer.set((id) => {
  trace('hold\n');
  Timer.schedule(id);
}, 5000);
Timer.schedule(holdTimer);

const button = new Digital({
  pin: 9,
  mode: Digital.Input,
  edge: Digital.Rising | Digital.Falling,

  onReadable() {
    if (this.read() === 0) {
      Timer.schedule(holdTimer, 2000);
    } else {
      Timer.schedule(holdTimer);
    }
  },
});
```

I'm not able to implement this with `setTimeout` and `clearTimeout` only. Something like `System.scheduleTimeout` would make sense i think.

phoddie, 2023-06-17:
If I understand correctly, the behavior you want to implement is a button press that triggers after two seconds (press-and-hold).

The timeout and invterval calls on `System` are intended to parallel the global functions of the same name (`setInterval`, etc) that are widely used on the web. The Moddable SDK Timer module has some additional features -- like initial interval on repeat and rescheduling . You are using reschedule here. Maybe that isn't necessary here? It is possible on the web to implement press-and-hold with the available APIs, so it should be here too.

Here's a revision of your example that just uses `System.setTimeout()` and `System.clearTimeout()`:

```js
const button = new Digital({
	pin: 9,
	mode: Digital.Input,
	edge: Digital.Rising | Digital.Falling,

	onReadable() {
		if (this.read() === 0) {
			this.timeout ??= System.setTimeout(() => {
				trace('hold\n');
				delete this.timeout;
			}, 2000);
		}
		else {
			System.clearTimeout(this.timeout);
			delete this.timeout;
		}
	}
});
```

FWIW -  this version keeps all of the code and configuration in the Digital constructor's options object. I prefer keeping everything together as a unit for maintenance and readability.

phoddie, 2023-06-17:
From your comment "working with some lousy push buttons" it occurred to me that maybe you want to debounce the buttons. If that's the case, here's a variation of the example above. It uses a simple debounce technique from the [Digital In node](https://github.com/phoddie/node-red-mcu/blob/e3e773cd8f3dae82cd0a617d4364de74186a4011/nodes/mcu/digital/digital.js#L57-L70) in Node-RED MCU Edition adapted from Node-RED.

```js
import Digital from 'embedded:io/digital';

const button = new Digital({
	pin: 9,
	mode: Digital.Input,
	edge: Digital.Rising | Digital.Falling,

	onReadable() {
		this.timeout ??= System.setTimeout(() => {
			delete this.timeout;
			if (0 === this.read())
				trace('pressed\n');
		}, 50);
	}
});
```

bartmichu, 2023-06-17:
@phoddie thank you so much. I was after the press-and-hold functionality without importing `Timer`. I came up with somewhat similar solution to your first snippet, but I was setting the timeout variable to `null` instead of using the `delete` operator and ended up with a lot of exceptions thrown at me. Your solution is very neat and works well, thanks!

closed 2023-06-17 by bartmichu

phoddie, 2023-06-17:
Glad that works for you.

FWIW – a useful way to think about the `Timer` module is that is a functional superset of the Web's timeout/interval APIs. It can be used to emulate those. It has some optimizations, like rescheduling and suspending, that are helpful in certain  scenarios, especially when embedded is pushing up against performance limits.