Moddable-OpenSource/moddable #1164: SPIFFS filesystem does not build for esp32-c3

tve, 2023-06-24:
**Build environment:** Linux
**Moddable SDK version:** 3.9.4
**Target device:** esp32c3

**Description**
Building a project that uses a spiffs filesystem does not work on the esp32-c3, it results in linker errors.

**Steps to Reproduce**
1. Run mcconfig on an app that includes `$(MODDABLE)/modules/files/file/manifest.json`
```
/home/tve/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: xs_esp32c3.a(modFile.c.o): in function `xs_file_iterator_destructor':
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:300: undefined reference to `closedir'
/home/tve/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: xs_esp32c3.a(modFile.c.o): in function `xs_File_Iterator':
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:333: undefined reference to `opendir'
/home/tve/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: xs_esp32c3.a(modFile.c.o): in function `xs_file_iterator_next':
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:368: undefined reference to `readdir'
/home/tve/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: xs_esp32c3.a(modFile.c.o): in function `xs_directory_create':
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:422: undefined reference to `mkdir'
/home/tve/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/8.4.0/../../../../riscv32-esp-elf/bin/ld: xs_esp32c3.a(modFile.c.o): in function `xs_directory_delete':
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:506: undefined reference to `rmdir'
```

phoddie, 2023-06-24:
Will take a look. My C3 is at the office, so need to get that first

phoddie, 2023-06-24:
I enabled the VFS support as you suggested above and successfully ran $MODDABLE/examples/files/file with `-p esp32/esp32c3`. Here's the instrumented output:

```
--- idf_monitor on /dev/cu.usbserial-110 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x48
load:0x403cc710,len:0x67c
load:0x403ce710,len:0x2260
entry 0x403cc710
I (104) cpu_start: Pro cpu up.
I (113) cpu_start: Pro cpu start user code
I (113) cpu_start: cpu freq: 160000000
I (113) cpu_start: Application information:
I (113) cpu_start: Project name:     xs_esp32
I (117) cpu_start: App version:      3.9.5-11-g524448b91-dirty
I (123) cpu_start: Compile time:     Jun 24 2023 12:13:46
I (128) cpu_start: ELF file SHA256:  87c7ebef7bf1e8ac...
I (133) cpu_start: ESP-IDF:          v4.4.3-dirty
I (137) heap_init: Initializing. RAM available for dynamic allocation:
I (144) heap_init: At 3FC8E030 len 0004E6E0 (313 KiB): DRAM
I (149) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (155) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (160) spi_flash: detected chip: generic
I (164) spi_flash: flash io: dio
I (167) sleep: Configure to isolate all GPIO pins in sleep state
I (172) sleep: Enable automatic swi??instruments key: Pixels drawn,Frames drawn,Network bytes read,Network bytes written,Network sockets,Timers,Files,Poco display list used,Piu command List used,SPI flash erases,Turns,System bytes free,CPU,Chunk used,Chunk available,Slot used,Slot available,Stack used,Stack available,Garbage collections,Keys used,Modules loaded,Parser used,Floating Point
This is a test.
We can write multiple values.
This is the end of the test.

test.txt renamed to test2.txt

name: Brian, city: Del Mar, state: CA

File length: 20
Last five shorts: 5 6 7 8 9 
test2.txt                        file          75 bytes
preferences.json                 file          46 bytes
test.bin                         file          20 bytes

Used/Total: 1506/52961

instruments: 1,0,0,0,0,1,0,0,0,0,0,215912,97,656,1536,1920,8176,1344,6144,2,1,1,0,3
```

tve, 2023-06-24:
Cool! I'm still stuck.. I did a fresh checkout of the moddable repo and when I build and run it just hangs at boot. Or do I need to pass some flags to mcconfig so I see output?
I'm running
```
mcconfig -i -m -p esp32/esp32c3
```

And I get:
```
--- idf_monitor on /dev/ttyACM0 115200 ---                                                      
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---                           
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x48 
load:0x403cc710,len:0x668
load:0x403ce710,len:0x2258
entry 0x403cc710
```
And it hangs there. I note that there are a couple bytes of differences compared to your binaries.
If I use ctrl-T ctrl-R to reset, I see:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x420378e4
0x420378e4: cpu_ll_waiti at /home/sw/moddable/esp-idf/components/hal/esp32c3/include/hal/cpu_ll.h:204
 (inlined by) esp_pm_impl_waiti at /home/sw/moddable/esp-idf/components/esp_pm/pm_impl.c:837

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x48
load:0x403cc710,len:0x668
load:0x403ce710,len:0x2258
entry 0x403cc710
```
git describe in the esp-idf dir says `v4.4.3`.
I haven't used instrumented builds, so perhaps I'm doing something wrong. Will test with a hello world main.
(I did do a flash erase and no change)

phoddie, 2023-06-24:
Im not sure it is hanging. I think there's just no output. On C3, it looks like instrumentation isn't configured to output to serial. I changed that earlier. Here's the modified `$MODDABLE/build/devices/esp32/xsProj-esp32c3/sdkconfig.inst` I'm using.

```
##### TEST INSTRUMENTATION
#
# ESP32-specific
#
CONFIG_ESP_COREDUMP_ENABLE_TO_UART=y

#
# ESP32-specific
#
CONFIG_ESP_CONSOLE_UART_DEFAULT=y
CONFIG_ESP_CONSOLE_UART_CUSTOM=n
CONFIG_ESP_CONSOLE_NONE=n
CONFIG_ESP_CONSOLE_UART_NUM=0

#
# Log output
#
CONFIG_LOG_DEFAULT_LEVEL_NONE=n
CONFIG_LOG_DEFAULT_LEVEL_ERROR=n
CONFIG_LOG_DEFAULT_LEVEL_WARN=n
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=n
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=n
CONFIG_LOG_DEFAULT_LEVEL=3

CONFIG_ESP_DEBUG_STUBS_ENABLE=y
CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=y
```

tve, 2023-06-24:
Alright. Hello World using an instrumented build now works for me, thanks for the help!
The file example still doesn't build, however:
```
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:300: undefined reference to `closedir'
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:333: undefined reference to `opendir'
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:368: undefined reference to `readdir'
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:422: undefined reference to `mkdir'
/home/sw/moddable/moddable/modules/files/file/esp32/modFile.c:506: undefined reference to `rmdir'
```
I changed:
```
(public) /h/s/m/m/b/m/lin> git diff ../../devices/esp32/xsProj-esp32c3/sdkconfig.defaults                            
diff --git a/build/devices/esp32/xsProj-esp32c3/sdkconfig.defaults b/build/devices/esp32/xsProj-esp32c3/sdkconfig.def
aults                                                                                                                
index ec81e3b7..b5627160 100644                                                                                      
--- a/build/devices/esp32/xsProj-esp32c3/sdkconfig.defaults                                                          
+++ b/build/devices/esp32/xsProj-esp32c3/sdkconfig.defaults                                                          
@@ -1155,8 +1155,8 @@ CONFIG_UNITY_ENABLE_IDF_TEST_RUNNER=y                                                          
 #                                                                                                                   
 # Virtual file system                                                                                               
 #                                                                                                                   
-CONFIG_VFS_SUPPORT_IO=n                                                                                             
-CONFIG_VFS_SUPPORT_DIR=n                                                                                            
+CONFIG_VFS_SUPPORT_IO=y                                                                                             
+CONFIG_VFS_SUPPORT_DIR=y                                                                                            
 CONFIG_VFS_SUPPORT_SELECT=n                                                                                         
 CONFIG_VFS_SUPPRESS_SELECT_DEBUG_OUTPUT=y                                                                           
 CONFIG_VFS_SUPPORT_TERMIOS=n                                                                                        
```
Setting _SELECT=y doesn't make it any better. I blew away build/tmp/esp32 and build/bin/esp32, no change. I'll have to dig further later unless you have the answer. A grep for VFS in all the sdconfig.defaults shows rather inconsistent results, of course, the IDF is probably inconsistent itself...

phoddie, 2023-06-24:
Right. As noted above, "I enabled the VFS support as you suggested." That was referring to your comment "Simply enabling _IO & _DIR... " That allows me to link. We don't use `select()` so there's no need to enable that.

phoddie, 2023-06-27:
The changes to allow it build are now available. Closing this out.

I understand that you may be having an issue at runtime. I wasn't able to reproduce that. If it persists, please open a separate issue with details so we can resolve that too. Thanks.

closed 2023-06-27 by phoddie