Moddable-OpenSource/moddable #1169: Docker container with moddable SDK and Espressif tools

profjmer, 2023-07-04:
I tried the procedures (manual and xs-dev) on Windows without success.
Could you build a container so that development is possible under Windows.
Thank you

phoddie, 2023-07-04:
> I tried the procedures (manual and xs-dev) on Windows without success.

Sorry to hear that. We'd really like to hear about any issues you ran into, so we can help you solve those and update the tools and instructions as needed to others don't have the same problem. For xs-dev, you can report [issues here](https://github.com/HipsterBrown/xs-dev/issues). @HipsterBrown usually replies quickly.

> Could you build a container so that development is possible under Windows.

The Stack-chan project has a [Docker image for Moddable SDK](https://github.com/meganetaaan/stack-chan/blob/bed4fc40cb237f703b44c95c55568070dfa6c944/firmware/docs/getting-started.md#using-docker-images-for-linux-only) builds. It is primarily intended for Linux but some users have had success using on Windows with WSL.

linfan68, 2023-08-04:
I’m mainly using WSL + Docker and esp-idf with the help of vscode as my dev environment. You’ll need this to forward USB devices into the container. https://learn.microsoft.com/en-us/windows/wsl/connect-usb