Moddable-OpenSource/moddable #1181: Run ESP32S3 on Windows

NextWorldLab, 2023-07-25:
Now that ESP32S3 and ESP32C3 are popular, I would like to use it on Windows as well.
I believe debugging is currently only available via USB-UART.
It would be nice to be able to debug using a USB-CDC such as M5ATOM_S3 or XIAO ESP32S3.

Currently, the response disappears on the following screen.
```
C:\pjt\moddable\examples\helloworld>mcconfig -d -m -p esp32/m5atom_s3
情報: 与えられたパターンのファイルが見つかりませんでした。
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf4.4_py3.8_env\Scripts\
Python 3.8.7
Using Git in C:\Espressif\tools\idf-git\2.34.2\cmd\
git version 2.34.1.windows.1
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v4.4.4

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.26.0.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.23.1\bin
    C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.3\ccache-4.3-windows-64
    C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
    C:\Espressif\frameworks\esp-idf-v4.4.4\tools

Checking if Python packages are up to date...
Python requirements from C:\Espressif\frameworks\esp-idf-v4.4.4\requirements.txt are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Recommend using ESP-IDF v4.4.3 (found v4.4.4)
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsproj-esp32s3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=c:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc -DSDKCONFIG_HEADER=c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=115200 -DIDF_TARGET=esp32s3 -DESP32_SUBCLASS=esp32s3 -DSDKCONFIG_DEFAULTS=c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3\sdkconfig.mc -DCCACHE_ENABLE=1 c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsproj-esp32s3"...
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: C:/Espressif/frameworks/esp-idf-v4.4.4/components/freemodbus.
Loading defaults file C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc...
C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1433 CONFIG_CONSOLE_UART_DEFAULT was replaced with CONFIG_ESP_CONSOLE_UART_DEFAULT
C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1434 CONFIG_CONSOLE_UART was replaced with CONFIG_ESP_CONSOLE_UART
C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1435 CONFIG_CONSOLE_UART_BAUDRATE was replaced with CONFIG_ESP_CONSOLE_UART_BAUDRATE
C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1438 CONFIG_FLASHMODE_DIO was replaced with CONFIG_ESPTOOLPY_FLASHMODE_DIO
C:\Users\tauchi\AppData\Local\Temp\confgen_tmp1zm582pw:504 line was updated to CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=n
C:\Users\tauchi\AppData\Local\Temp\confgen_tmp1zm582pw:585 line was updated to CONFIG_ESP_COREDUMP_ENABLE_TO_UART=n
C:\Users\tauchi\AppData\Local\Temp\confgen_tmp1zm582pw:590 line was updated to CONFIG_ESP_COREDUMP_ENABLE=n
-- Configuring done
-- Generating done
-- Build files have been written to: C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/build
        1 個のファイルをコピーしました。
# ld xs_esp32.bin
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
Executing action: all (aliases: build)
Running ninja in directory c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsproj-esp32s3\build
Executing "ninja all"...
[... 73 lines cut ...]
Writing at 0x0006683e... (82 %)
Writing at 0x0006f0aa... (88 %)
Writing at 0x00076bf0... (94 %)
Writing at 0x0007c5d7... (100 %)
Wrote 459280 bytes (272177 compressed) at 0x00010000 in 3.8 seconds (effective 979.1 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.1 seconds (effective 224.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
Launching app...
Type Ctrl-C twice after debugging app.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x20 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a79
waiting for download
```
xsbug started but not working

Thank you.,

mkellner, 2023-07-25:
Press reset on the device after the software downloads.

NextWorldLab, 2023-07-26:
thank you for your reply.
Debug instrument does not start even after writing and resetting.

The last part of the message is
The last 9 lines are the message after reset.
```
Writing at 0x00076bf0... (94 %)
Writing at 0x0007c5d7... (100 %)
Wrote 459280 bytes (272177 compressed) at 0x00010000 in 3.7 seconds (effective 981.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 83...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.1 seconds (effective 223.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
Launching app...
Type Ctrl-C twice after debugging app.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x20 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375418
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x88
load:0x403c9700,len:0x990
load:0x403cc700,len:0x2700
entry 0x403c98ac
```
the environment is
・Windows 11
・Platform esp32/ｍ5atom_s3, esp32/xiao_esp32s3

I thought I still couldn't debug because the Windows Makefile didn't recognize the USB mode.
tools/mcconfig/nmake.esp32.mk
Mac and Linux Makefiles use the VID, PID, and USE_USB flags to control flow.
tools/mcconfig/make.esp32.mk

thank you.

NextWorldLab, 2023-08-16:
There are two issues in my opinion.

I am using Windows10, M5Atom_S3.

1. After flashing the device and resetting it, the VID/PID is 303a/1001 and it won't go into CDC mode.
2. macで焼いたデバイスを接続してもxsbugに接続できない。使用したコマンドは”mcconfig -d -m -p esp32/m5atom_s3 -t xsbug”

mkellner, 2023-08-19:
Hello @NextWorldLab, I have checked in some support to nmake.esp32.mk
Please try it.

NextWorldLab, 2023-08-20:
@mkellner  san

I confirmed the operation with the device (M5atom_s3).
Now I can use the ESP32S3 device.

Thank you,

phoddie, 2023-08-20:
That's great news. Thank you for verifying the fix. Should we close this issue now?

NextWorldLab, 2023-08-21:
NodeRED Editor and xsbug cannot be connected with Node-RED MCU Plugin.
I could be wrong, but I have a feeling that mcconfig's x option is not working.
It works fine for m5atom_matrix(ESP32) but not for m5atom_s3(esp32_s3).

phoddie, 2023-08-21:
Thanks for the report.

I think you are correct: it looks like the [USB cases in nmake.esp32.mk](url) do not set `XSBUG_HOST` and `XSBUG_PORT`. That should be straightforward to fix. I'll check with @mkellner about that.

mkellner, 2023-08-21:
Hello @NextWorldLab, please try again.

phoddie, 2023-08-21:
Just FYI – here's the relevant [commit](https://github.com/Moddable-OpenSource/moddable/commit/e7d90e7e8ae9dab171cf36591a6ae105d8357ad0).

NextWorldLab, 2023-08-22:
thank you.
Debugging is now available in the NodeRED MCU Plugin.
I will close.

In addition, although it is almost okay, it may not be possible to connect to xsbug in the first Build. Still, the second time (-t xsbug) is fine.
Even with the same code, the reproducibility is poor, so I will write it again in ISSUE when I understand the reproduction procedure.
I have experienced
example/helloworld is fine.
example/piu/balls I had a bad time and it doesn't reproduce.
example/piu/text is fine.

closed 2023-08-22 by NextWorldLab

phoddie, 2023-08-22:
> Debugging is now available in the NodeRED MCU Plugin.

Good news.

>  it may not be possible to connect to xsbug in the first Build. Still, the second time (-t xsbug) is fine.
> so I will write it again in ISSUE when I understand the reproduction procedure.

That's strange. Please do report the details when you know more. Thank you.