Moddable-OpenSource/moddable #1196: granel

granel2, 2023-08-19:
Do you have questions about

- an example
- where to find documentation
- locating the source code for a specific module
- anything else about the Moddable SDK?

You are in the wrong place!

Here are some good places to start:

- Check out [Discussions](https://github.com/Moddable-OpenSource/moddable/discussions) to see if other developers are asking the same question.
- Search through the [Issues already created](https://github.com/Moddable-OpenSource/moddable/issues?q=is%3Aissue+). Maybe your question has come up before.
- Read the [Documentation](https://github.com/Moddable-OpenSource/moddable/tree/public/documentation#moddable-sdk--documentation-overview). There's lots of detailed documentation of a wide range of topics.
- Ask a question in our [chatroom on Gitter](https://gitter.im/embedded-javascript/moddable).

If you want to ask your question here on GitHub, please ask it using [Discussions](https://github.com/Moddable-OpenSource/moddable/discussions). Questions opened as an issue will be converted to a Discussion.

phoddie, 2023-08-20:
Welcome.

A great alternative to the manual set-up is to use [xs-dev](https://xs-dev.js.org). It is a tool that takes automates the installation and set-up on the Moddable SDK. If you have npm installed, here's all you have to do:

```
npm install -g xs-dev
xs-dev setup
```

Good luck!

NextWorldLab, 2023-08-20:
Does the xsbug.exe file exist in the %MODDABLE%\build\bin\win\release folder?
If so, is the environment variable (path) set? (Launch the Windows terminal after setting it. Do not use a new tab.)

granel2, 2023-08-20:
but I read in the manual xs-dev -- this installation option does not work with Windows, but I started with it, the macbook is waiting) - if I do the installation with the xs-dev option, I need to clean the computer from what I tried with manual installation == from a clean leaf?

granel2, 2023-08-20:
about : Does the xsbug.exe file exist in the %MODDABLE%\build\bin\win\release folder?
in build folder have only folders:
device; makefile; simulators
(((

granel2, 2023-08-20:
I did the installation on Windows 11-64 strictly according to the instructions for manual installation - I never managed to create a folders bin\release  -  have only  C:\Users\admin\Projects\moddable\build\  inside 3 other folders: device + makefile + simulators

granel2, 2023-08-20:
installation attempts with xs-dev end up with this message when trying to run a test  project:

C:\Users\admin\Projects\moddable>xs-dev run --example helloworld
Moddable tooling required. Run 'xs-dev setup --device win' before trying again.

I install this:

C:\Program Files\Microsoft Visual Studio\2022\Community>xs-dev setup --device win
Setting up Windows tools at C:\Users\admin\Projects\moddable
......................
.....................
The Moddable Command Prompt will now open for you in a new Window. Please close this Command Prompt and use the Moddable Command Prompt instead.
As a next step, try running the "helloworld example" in the Moddable Command Prompt: xs-dev run --example helloworld'
Opening the Moddable Command Prompt in a new Window

bud all bad((

granel2, 2023-08-20:
my attempts for 3 days to install and start working with moddable sdk + esp and start learning java script - are not successful, probably I need to return to my favorite C language and development tools - too much time wasted just to install the environment on Win11 and MacOS . The tips found for installation on the Internet are in the nature of a secret society of gurus)))

phoddie, 2023-08-20:
Sorry that you weren't able to get things working and the frustration it caused. I think both @NextWorldLab and I have been guessing about how to help you, because there haven't been many details in the reports here beyond "does not work."

@HipsterBrown has been very successful helping developers through any problems using xs-dev. If you do decide to try again, I expect he would be able to suggest some ideas for tracking down what is going wrong on your setup.

closed 2023-08-20 by phoddie

granel2, 2023-08-20:
@HipsterBrown xs-dev + my frend - (working with js - writing servers software) help me instal node.js on my macbook (gave me commands that were not on the Internet) -- but what to do next? install moddableSDK? how to do examples code on my pc? how dounload to esp-esp32-pico-...? I really need step-by-step instructions for the first test project - if it succeeds, then everything will be clear

granel2, 2023-08-20:
need or no install xcode? when install esp sdk?

granel2, 2023-08-20:
for a start, you need minimal help - in simple words it will explain what and in what sequence to install (maybe - where to find it)
have macos 11.7.9 -- for xcode 14 need  >ios13 (((

granel2, 2023-08-21:
ones more try install moddable win11:

C:\Users\admin\Projects\examples\ch1-gettingstarted\host>mcconfig -d -m -p esp
'mcconfig' is not recognized as an internal or external command,
operable program or batch file.

where is and how install -- mcconfig
??

granel2, 2023-08-21:
can see:

C:\Users\admin\Projects\moddable\tools\mcconfig

HipsterBrown, 2023-08-21:
@granel2 have you read through the starting guide on the xs-dev docs? https://xs-dev.js.org/guide/00-prepare/

This walks through the initial system requirements, how to install the CLI, and how to start using it.

granel2, 2023-08-21:
I made a new attempt at a standard installation on Windows 11 - I went as far as creating an example and failed again. Now I'll try a fresh installation of Win11 from your link.

granel2, 2023-08-21:
√ Moddable SDK successfully set up!
it's with xs-dev win11

granel2, 2023-08-21:
C:\Users\admin\Projects\moddable>xs-dev run --example helloworld
Moddable tooling required. Run 'xs-dev setup --device win' before trying again.
//installing win

C:\Users\admin\Projects\moddable>xs-dev setup --device win
Setting up Windows tools at C:\Users\admin\Projects\moddable
i Moddable repo already installed
√ Creating Moddable SDK Environment Batch File
√ Creating Moddable Command Prompt Shortcut
√ Installing xsbug-log dependencies
√ Moddable SDK successfully set up!
A shortcut to the Moddable Command Prompt has been created at C:\Users\admin\xs-dev\Moddable Command Prompt.lnk.
You should always use the Moddable Command Prompt when working with the Moddable SDK.
The Moddable Command Prompt will now open for you in a new Window. Please close this Command Prompt and use the Moddable Command Prompt instead.
As a next step, try running the "helloworld example" in the Moddable Command Prompt: xs-dev run --example helloworld'
Opening the Moddable Command Prompt in a new Window

but in new window same : 

C:\Users\admin\Projects\moddable>xs-dev run --example helloworld
Moddable tooling required. Run 'xs-dev setup --device win' before trying again.

granel2, 2023-08-21:
I stopped at this and on the previous installation attempt

granel2, 2023-08-21:
one step left before my victory

phoddie, 2023-08-21:
Thank you for trying again and posting the details from the run of xs-dev. It looks like, for some reason, xs-dev doesn't see the installed files in the new window. Let's see if @HipsterBrown has any suggestions for how to investigate what is going on.

granel2, 2023-08-21:
ok, try now

granel2, 2023-08-21:
I can't solve the problem myself((((

C:\Users\admin\Projects\moddable>xs-dev run --example helloworld
Moddable tooling required. Run 'xs-dev setup --device win' before trying again.

phoddie, 2023-08-21:
Yes, I understand. Please wait a little to see if we can get you a suggestion to try. Thank you.

granel2, 2023-08-21:
ok))))))))))))

andycarle, 2023-08-21:
Could you please post a screenshot of the Command Prompt window where you're running mcconfig? My guess is that you are not using the Moddable Command Prompt that is installed by xs-dev.

granel2, 2023-08-21:
I tried to run in different terminals, even Power Shell)

andycarle, 2023-08-21:
Interesting. I haven't tried running xs-dev in an Administrator Command Prompt before, but I don't see any obvious reason that should be a problem. 

Could you share the output of the `env` command? I'm just wondering if the Moddable Command prompt batch file is successfully setting up the environment.

granel2, 2023-08-21:
[image]

granel2, 2023-08-21:
command ? :     
env ?

granel2, 2023-08-21:
[image]

granel2, 2023-08-21:
this?

granel2, 2023-08-21:
[image]

granel2, 2023-08-21:
[image]

andycarle, 2023-08-21:
Sorry, the easiest way to get what we need would be to just run 'set' with no arguments at the Moddable Command Prompt.

granel2, 2023-08-21:
try

granel2, 2023-08-21:
'xs-dev setup '==??

granel2, 2023-08-21:
[image]

granel2, 2023-08-21:
after cmd 'set' received big information)
[image]

andycarle, 2023-08-21:
If you could just copy and paste everything that `set` output, that would be great. (Ideally code formatted here so that it's legible.

granel2, 2023-08-21:
[set.txt](https://github.com/Moddable-OpenSource/moddable/files/12399439/set.txt)

granel2, 2023-08-21:
I think it full

andycarle, 2023-08-21:
Thanks. That output rules out the most obvious possible issues. I'll need to take a deeper look at this, which I won't be able to do until later tonight.

If anyone else has any thoughts, I'd love to hear them! Otherwise I'll keep digging late tonight.

granel2, 2023-08-21:
I really hope for your success! I'm waiting for my start of work - the hardware is ready! then I plan to master this environment for a macos.

granel2, 2023-08-21:
I'll probably start preparing the installation for macos, while the Gurus are thinking))

granel2, 2023-08-21:
start testing on macbook esp8288 - i think all ok

granel2, 2023-08-21:
sergei@MacBook-Pro ~ % sudo xs-dev run --example helloworld --device=esp8266 --port=/dev/tty.usbserial-1420
⠴ Building and deploying project /Users/sergei/.local/share/moddable/examples/helloworld on esp
⠏ Building and deploying project /Users/sergei/.local/share/moddable/examples/helloworld on esp

Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 5c:cf:7f:82:e7:a6
⠋ Building and deploying project /Users/sergei/.local/share/moddable/examples/helloworld on esp
Running stub...
Stub running...
Changing baud rate to 921600
⠙ Building and deploying project /Users/sergei/.local/share/moddable/examples/helloworld on esp
⠧ Building and deploying project /Users/sergei/.local/share/moddable/examples/helloworld on esp

A fatal error occurred: Timed out waiting for packet content
sergei@MacBook-Pro ~ % 
==============
I think - need download FW to esp8266 before? how it to do? and need correct serial port speed?

phoddie, 2023-08-21:
The `xs-dev run` builds, downloads, and launches the firmware. The error you show is coming from the Espressif esptool.py that does the flashing. We know it was able to communicate with the chip because we see `Chip is ESP8266EX`. That communication is done at a lower speed (115200 baud). Occasionally higher baud rates fail -- because of a bad USB cable or because your ESP8266 board does not support higher speeds. You might try this in the terminal on macOS to force the slower baud rate.

```
UPLOAD_SPEED=115200 xs-dev run --example helloworld --device=esp8266 --port=/dev/tty.usbserial-1420
```

If that works, you could try 46800 instead of 115200.

andycarle, 2023-08-22:
A quick question on the Windows issue: after you've run `xs-dev setup`, what are the contents of `C:\Users\admin\Projects\moddable\build\bin\win\release`?

granel2, 2023-08-22:
on the Windows issue: after you've run xs-dev setup, what are the contents of

[image]

granel2, 2023-08-22:
maybe I violated the order of installation of software modules? then I have to delete everything and try the installation for Win11 a third time?

NextWorldLab, 2023-08-22:
The environment (file location) used by xs-dev can be found with the "xs-dev dr" command.

NextWorldLab, 2023-08-22:
By the way, in x86native tool comand prompt for vs2021
> cd \Users\admin\Projects\moddable\build\
> build clean
>build
Do you get an error when you run

granel2, 2023-08-22:
[image]

granel2, 2023-08-22:
[image]

NextWorldLab, 2023-08-22:
The "xs-dev dr" command gives an error due to the python environment.
Is there anything in the %USERPROFILE%\.share\local folder?
Is moddable (used by xs-dev) included here?

granel2, 2023-08-22:
but after 'build' - see big work)

[image]

NextWorldLab, 2023-08-22:
sorry.
Since the MODDABLE environment variable is defined, it will be C:\Users\admin\Projects\moddable.

NextWorldLab, 2023-08-22:
build takes a long time.

granel2, 2023-08-22:
cd \Users\admin\Projects\moddable
build clean
build
#need to do this?

granel2, 2023-08-22:
now see ( after : cd \Users\admin\Projects\moddable\build\makefiles\win
build clean
build)

[image]

[67 more replies]