Moddable-OpenSource/moddable #1201: Should examples/drivers/sensors include manifest_base?

stc1988, 2023-08-27:
**Build environment:** macOS
**Moddable SDK version:** 4.0.0-23-g2f9eb25dc
**Target device:** esp32/m5stack
**Steps to Reproduce**

```
cd $MODDABLE/examples/drivers/sensors/sht3x
mcconfig -d -m -p esp32/m5stack -t build
### Error: 'undefined/network/wifi/manifest.json': manifest not found!
```

**Other information**
I think this issue is affected  by https://github.com/Moddable-OpenSource/moddable/commit/e8375250772614e4910211c78b005d97cc733c82.

Including `"(MODDABLE)/examples/manifest_base.json` make it works.

phoddie, 2023-08-27:
Yes!

There was a mistake in $MODDABLE/modules/io/manifest.json. It included manifest_base.json, but it should not. Only the project manage should do that. We corrected this in the nRF52 / Moddable Four work. Unfortunately, it caused problems with examples that depended on io/manifest.json to include manifest_base.json. I'll review examples/drivers/sensors and examples/drivers/rtc to add manifest_base.json where needed.

Sorry for the trouble. Thanks for the report and precise analysis.

phoddie, 2023-09-11:
Closing as this should now be resolved. If there are remaining issues, please re-open. Thank you.

closed 2023-09-11 by phoddie